smartdevicelink/sdl_core #2600: POssible race conditions in MessageLoopThread

LuxoftAKutsan, 2018-09-10:
In `MessageLoopThread<Q>::WaitDumpQueue()  ` exist possibility of race condition of work with message_queue_. 
In MessageLoopThread should be added additional synchronization primitives

jordynmackool, 2021-09-15:
@LuxoftAKutsan is this issue still present?

AKalinich-Luxoft, 2021-09-15:
@jordynmackool I think this issue is not actual for the current release as `MessageQueue<T, Q>` class has its internal `queue_lock_` to make everything synchronized inside.
Additionally, some stabilization jobs were done in #3471 so there is nothing to fix at the moment.

jordynmackool, 2021-09-15:
@AKalinich-Luxoft thanks for the quick response, can we close this issue based on your comment?

AKalinich-Luxoft, 2021-09-15:
@jordynmackool yes, this can be closed

jordynmackool, 2021-09-15:
Closing based on this [comment](https://github.com/smartdevicelink/sdl_core/issues/2600#issuecomment-920327308).

closed 2021-09-15 by jordynmackool