smartdevicelink/sdl_core #2603: Make Size() function thread safe to avoid core dump

HSavynetska, 2018-09-11:
### Bug Report
Make Size() function thread safe to avoid core dump 

##### Expected Behavior
The Size() function calls the CheckSetSizes() function which compares the sizes of the two sets and checks this value in DCHECK. So must be sure that the function Size() will not be called at the same time when there is addition or removal of requests from sets

##### Observed Behavior

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

HSavynetska, 2018-09-14:
Fixed by #1569

t-yoshii, 2018-10-02:
Hi, I also hit this issue. 
Are there any reasons or risks that this pull request (#1569) has not yet been merged?

LitvinenkoIra, 2018-10-08:
@jacobkeeler please take a look at https://github.com/smartdevicelink/sdl_core/pull/1569 one more time

theresalech, 2018-10-09:
@HSavynetska please provide Observed Behavior and OS & Version Information for this issue

LuxoftAKutsan, 2018-10-09:
@HSavynetska  Please make proper description . Not clear Size function of what class. 
If it is not just potential but fix some wrong behavior observer, please add it to the description

LitvinenkoIra, 2018-10-12:
closed as https://github.com/smartdevicelink/sdl_core/pull/1569 merged

closed 2018-10-12 by LitvinenkoIra