smartdevicelink/sdl_core #2612: SDL doesn't send OnHMIStatus(FULL) to the non-media App if OnEventChanged(AUDIO_SOURCE:true) was before.

Itileda, 2018-09-14:
### Bug Report
SDL doesn't send OnHMIStatus(FULL) to the non-media App if OnEventChanged(AUDIO_SOURCE:true) was before.

##### Reproduction Steps
1. Consent device
2. Send from HMI OnEventChanged(AUDIO_SOURCE:true)
3. Register non-media App

##### Steps
1. Activate App
##### Expected Behavior
SDL activates App, sends FULL notification

##### Observed Behavior
SDL doesn't send FULL to the App

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

HSavynetska, 2019-11-26:
The issue is relevant on the current develop branch

Environment:
 Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2064
 SDL build: [https://github.com/smartdevicelink/sdl_core/tree/develop] ( 891497b )
 ATF: develop branch

dboltovskyi, 2020-06-15:
Issue is very similar to https://github.com/smartdevicelink/sdl_core/issues/1908

jacobkeeler, 2020-07-27:
Closing as duplicate of #1908

closed 2020-07-27 by jacobkeeler