smartdevicelink/sdl_core #2633: Audio Stream cutoff when using pipe streaming

jacobkeeler, 2018-09-25:
### Bug Report
Audio stream data is cut off when using gstreamer to read from a pipe stream.

##### Reproduction Steps
1. Set `AudioStreamConsumer` to `pipe` in `smartDeviceLink.ini`.
2. Start audio stream from app (accept prompt from HMI to start audio stream).
3. Start reading audio stream using the following `gstreamer` command:
```
gst-launch-1.0 filesrc location=audio_stream_pipe ! audio/x-raw,format=S16LE,rate=16000,channels=1 ! pulsesink
```
4. Send an audio file longer than 1 minute over the Audio service

##### Expected Behavior
Entire audio file is played

##### Observed Behavior
Audio file stops playing after 15 seconds or so

The same does not happen when using socket streaming

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 4.5.1
* Testing Against: SPT 20180821-Nightly-Android

cat-fang, 2018-12-10:
Hello, was it solved?

theresalech, 2019-08-29:
Closed via #2956

closed 2019-08-29 by theresalech