smartdevicelink/sdl_core #2643: when SDL receive OnInteriorVehicleData, it shall update (not replace) the cached data

yang1070, 2018-09-28:
### Bug Report
when SDL receives OnInteriorVehicleData regarding LIGHT module, it shall update (not replace) the cached data. 
SDL shall replace cached data when receives a response of GetInteriorVehicleData.

##### Reproduction Steps
1. build sdl core release/5.0.0 and clone sdl hmi develop branch
2. run sdl core and sdl hmi
3. register a remote control app connect to sdl
4. activate the app from hmi
5. from the app, send GetInteriorVechicleData with moduleType=LIGHT, it get status of all available lights
6. on hmi change the status of one light
7. from the app, send GetInteriorVechicleData with moduleType=LIGHT again

##### Expected Behavior
App shall get t the status of all available lights
SDL shall send the status of all available lights in the response

##### Observed Behavior
App only get one light in the response.
SDL only include the status of one recent changed light in the response

##### OS & Version Information
* OS/Version: Ubuntu 16.04 
* SDL Core Version:  5.0 release candidate (release 5.0.0 branch)
* SDL HMI Version: current develop branch
* Testing Against: remote control of Light

jacobkeeler, 2018-10-11:
Closing with the merge of #2667

closed 2018-10-11 by jacobkeeler

yang1070, 2018-10-15:
@jacobkeeler 
I can produce it on branch release/5.0.0 with commit  up to c9697970a6ad6300936fe893efe7f3acae92a7c0

yang1070, 2018-10-15:
@JackLivio I have update my core to the latest commit, can still reproduce this issue

PR #2667 (2018-10-11): Merge module data instead of outright replacing each element

Fixes #2643 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
1. Subscribe to LIGHT module
2. Update one light status via the HMI
3. Send another GetInteriorVehicleData for LIGHT module with subscribe=true
4. Verify that all light values are present, including the updated value

### Summary
Merges module data in cache when an OnInteriorVehicleData is received instead of replacing each element in the root object.

### Changelog

##### Bug Fixes
* Merges interior vehicle cache data instea [... cut]