smartdevicelink/sdl_core #2653: build error due to wrong data types passed to GetDataOnSessionKey

mirzak, 2018-10-04:
### Bug Report

I see the following build error using 7f7fcbb998fb17f2954fd103349af67ea9b71a3f on master branch.

```
09:27:43.949 | /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/media_manager/src/audio/a2dp_source_player_adapter.cc: In member function 'virtual void media_manager::A2DPSourcePlayerAdapter::StartActivity(int32_t)':
09:27:43.949 | /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/media_manager/src/audio/a2dp_source_player_adapter.cc:95:79: error: no matching function for call to 'protocol_handler::SessionObserver::GetDataOnSessionKey(int32_t&, int, NULL, uint32_t*)'
09:27:43.949 |      session_observer_.GetDataOnSessionKey(application_key, 0, NULL, &device_id);
09:27:43.949 |                                                                                ^
09:27:43.949 | In file included from /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/connection_handler/include/connection_handler/connection_handler_impl.h:42:0,
09:27:43.949 |                  from /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/media_manager/include/media_manager/audio/a2dp_source_player_adapter.h:40,
09:27:43.949 |                  from /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/media_manager/src/audio/a2dp_source_player_adapter.cc:38:
09:27:43.949 | /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/include/protocol_handler/session_observer.h:241:19: note: candidate: virtual int32_t protocol_handler::SessionObserver::GetDataOnSessionKey(uint32_t, uint32_t*, std::__cxx11::list<int>*, transport_manager::DeviceHandle*) const
09:27:43.949 |    virtual int32_t GetDataOnSessionKey(
09:27:43.949 |                    ^~~~~~~~~~~~~~~~~~~
09:27:43.949 | /var/lib/go-agent/pipelines/GDP-Yocto-QEMU_x86-64-GITHUB-PR/gdp-src-build/tmp/work/core2-64-poky-linux/sdl-core/4.5.1-r0/git/src/components/include/protocol_handler/session_observer.h:241:19: note:   no known conversion for argument 4 from 'uint32_t* {aka unsigned int*}' to 'transport_manager::DeviceHandle* {aka long unsigned int*}'
```

##### Expected Behavior

Build passes.

##### OS & Version Information
* OS/Version: Yocto Build environment (sumo, gcc7, glibc2.26)

##### Test Case, Sample Code, and / or Example App

This patch workarounds/fixes the issue I am having

https://gist.github.com/mirzak/f824ebfe4e8ef0a9eb235beacdb18bb1

jacobkeeler, 2018-10-04:
Duplicate of #2137 (fixed on develop)

closed 2018-10-04 by jacobkeeler

mirzak, 2018-10-04:
Not the same error, thought I did initially come to the same conclusion.