smartdevicelink/sdl_core #2661: SDL 0188 - Interior Vehicle Data Resumption

jordynmackool, 2018-10-09:
Proposal: [Interior Vehicle Data Resumption](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0199-Adding-GPS-Shift-support.md)

> This proposal is to restore Interior data subscriptions on application registration after unexpected disconnect or ignition cycle.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/554

Steering Committee Decision:

> The Steering Committee accepted this proposal with revisions on 8/1/2018. Revisions were made on 10/9/18.

theresalech, 2020-08-26:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0188 Revisions - Interior Vehicle Data resumption](https://github.com/smartdevicelink/sdl_evolution/issues/1072).  Revisions include fixing parameter names for `GetInteriorVehicleData` from `IsSubscribe` to `subscribe` according to the existing API, and replacing `SetInteriorVehicleData` with `GetInteriorVehicleData` in the [Behavior changes](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0188-get-interior-data-resumption.md#getinteriorvehicledata-behavior-changes) section of the proposal's `Proposed solution`. Please reference the issue and proposal markdown file for revisions details.

jacobkeeler, 2020-09-04:
Closing with the merge of #3480

closed 2020-09-04 by jacobkeeler