smartdevicelink/sdl_core #2662: PerformAudioPassThru RPC ignores bit rate and sample rate

jyh947, 2018-10-09:
### Bug Report
After sending a PerformAudioPassThru RPC to SDL core from the Ford SYNC 3 emulator, SDL core seems to discard the bit rate, sample rate, and the audio type.

##### Reproduction Steps
1. Launch the Ford SYNC 3 8" emulator
2. Send a PerformAudioPassThru with 16kHz sample rate and 16 bits/sample.
3. Open returned PCM audio file in your favorite audio editing program and specify your audio bit rate and sample rate as 16kHz at 16 bits/sample.

##### Expected Behavior
PCM audio sounds like what was recorded

##### Observed Behavior
PCM audio sounds distorted.  This is because the audio was not properly recorded at 16kHz at 16 bits/sample.  The actual PCM audio is actually at a 44.1kHz sample rate.

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: The version included with the SYNC 3 emulator.  (It should be noted that in the code section, I will be referring to the current master branch of the SDL Core repo)
* Testing Against: Core

##### Test Case, Sample Code, and / or Example App
PerformAudioPassThru calls the following function and passes in all the RPC fields from the emulator
```
void PerformAudioPassThruRequest::StartMicrophoneRecording() {
  LOG4CXX_AUTO_TRACE(logger_);

  uint32_t app_id = connection_key();
  application_manager_.BeginAudioPassThru(app_id);

  application_manager_.StartAudioPassThruThread(
      connection_key(),
      correlation_id(),
      (*message_)[str::msg_params][str::max_duration].asInt(),
      (*message_)[str::msg_params][str::sampling_rate].asInt(),
      (*message_)[str::msg_params][str::bits_per_sample].asInt(),
      (*message_)[str::msg_params][str::audio_type].asInt());
}
```
The application manager takes in all these fields, but doesn't do anything with 3 of the fields: `sampling_rate`, `bits_per_sample`, `audio_type`.
```
void ApplicationManagerImpl::StartAudioPassThruThread(int32_t session_key,
                                                      int32_t correlation_id,
                                                      int32_t max_duration,
                                                      int32_t sampling_rate,
                                                      int32_t bits_per_sample,
                                                      int32_t audio_type) {
  LOG4CXX_AUTO_TRACE(logger_);
  LOG4CXX_INFO(logger_, "START MICROPHONE RECORDER");
  DCHECK_OR_RETURN_VOID(media_manager_);
  media_manager_->StartMicrophoneRecording(
      session_key, get_settings().recording_file_name(), max_duration);
}
```
The `SamplingRate` and `AudioCaptureQuality` enums are properly defined, but are unused.
https://github.com/smartdevicelink/sdl_core/blob/7f7fcbb998fb17f2954fd103349af67ea9b71a3f/src/components/media_manager/include/media_manager/audio/audio_stream_sender_thread.h

shoamano83, 2018-10-16:
Just FYI, I have opened a PR to fix the parameter handlings in Core:
https://github.com/smartdevicelink/sdl_core/pull/2184

I do not know if that PR also fixes SYNC 3 emulator since I believe it is proprietary.

jyh947, 2018-10-17:
I built this version of core with the latest release of the Ford SYNC 3 emulator and this PR does not solve the PerformAudioPassThru issue.

Here is what I'm getting from the debug log with the branch from your build:

```
DEBUG [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/rpc_handler_impl.cc:204 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: 			Message to convert: protocol 5; json {"audioPassThruDisplayText1":"APT Display Text 1","maxDuration":10000,"audioPassThruDisplayText2":"APT Display Text 2","samplingRate":"8KHZ","muteAudio":true,"bitsPerSample":"8_BIT","audioType":"PCM"}
DEBUG [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/rpc_handler_impl.cc:242 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Convertion result for sdl object is true function_id 16
DEBUG [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/rpc_handler_impl.cc:345 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Successfully parsed message into smart object
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/rpc_handler_impl.cc:200 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Exit
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RPCServiceImpl] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/rpc_service_impl.cc:63 rpc_service::RPCServiceImpl::ManageMobileCommand: Enter
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][ApplicationManager] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/application_manager_impl.cc:2739 ApplicationManagerImpl::IsLowVoltage: result: 0
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][ApplicationManager] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/application_manager_impl.cc:3147 ApplicationManagerImpl::IsAppInReconnectMode: Enter
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][ApplicationManager] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/application_manager_impl.cc:3147 ApplicationManagerImpl::IsAppInReconnectMode: Exit
TRACE [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][ApplicationManager] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/src/application_manager_impl.cc:3147 ApplicationManagerImpl::IsAppInReconnectMode: Exit
WARN  [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][VehicleInfoPlugin] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/rpc_plugins/vehicle_info_plugin/src/vehicle_info_mobile_command_factory.cc:139 CommandCreator& vehicle_info_plugin::VehicleInfoMobileCommandFactory::buildCommandCreator: Unsupported function_id: 16
DEBUG [17 Oct 2018 14:03:52,577][0x7f6aaeffd700][RemoteControlModule] /home/jeffrey/Downloads/sdl_core-6a830e1cb89cad42167fcaa22af9765ce8c085ba/src/components/application_manager/rpc_plugins/rc_rpc_plugin/src/rc_command_factory.cc:220 CommandCreator& rc_rpc_plugin::RCCommandFactory::get_mobile_creator_factory: CreateMobileCommand function_id: 16 message_type: -1
```

Here is the debug output from the 5.0.0 RC2 version of SDL core:

```
DEBUG [12 Oct 2018 08:01:54,084][0x7fb9727fc700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/rpc_handler_impl.cc:235 rpc_handler::RPCHandlerImpl::ConvertMessageToSO:                  Message to convert: protocol 5; json {"audioPassThruDisplayText1":"APT Display Text 1","maxDuration":10000,"audioPassThruDisplayText2":"APT Display Text 2","samplingRate":"16KHZ","muteAudio":true,"bitsPerSample":"16_BIT","audioType":"PCM"}
DEBUG [12 Oct 2018 08:01:54,084][0x7fb9727fc700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/rpc_handler_impl.cc:286 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Convertion result for sdl object is true function_id 16
DEBUG [12 Oct 2018 08:01:54,084][0x7fb9727fc700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/rpc_handler_impl.cc:389 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Successfully parsed message into smart object
TRACE [12 Oct 2018 08:01:54,084][0x7fb9727fc700][RPCHandlerImpl] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/rpc_handler_impl.cc:231 rpc_handler::RPCHandlerImpl::ConvertMessageToSO: Exit
TRACE [12 Oct 2018 08:01:54,084][0x7fb9727fc700][RPCServiceImpl] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/rpc_service_impl.cc:63 rpc_service::RPCServiceImpl::ManageMobileCommand: Enter
TRACE [12 Oct 2018 08:01:54,084][0x7fb9727fc700][ApplicationManager] /home/jeffrey/Downloads/sdl_core-5.0.0_RC2/src/components/application_manager/src/application_manager_impl.cc:2582 ApplicationManagerImpl::IsLowVoltage: Result: 0
[... cut ...] rintSmartObject: SMART OBJECT: {
   "msg_params" : {
      "audioPassThruDisplayText1" : "APT Display Text 1",
      "audioPassThruDisplayText2" : "APT Display Text 2",
      "audioType" : 0,
      "bitsPerSample" : 1,
      "maxDuration" : 10000,
      "muteAudio" : true,
      "samplingRate" : 1
   },
   "params" : {
      "connection_key" : 65537,
      "correlation_id" : 17,
      "function_id" : 16,
      "message_type" : 0,
      "protocol_type" : 0,
      "protocol_version" : 5
   }
}
```

ShobhitAd, 2021-06-01:
Unable to reproduce the issue on the latest develop (https://github.com/smartdevicelink/sdl_core/commit/9ac9acdec6d82b280fca5286549ea904e7bc8b4c)

closed 2021-06-01 by ShobhitAd