smartdevicelink/sdl_core #2663: if there is no change on module data, SDL/HMI shall not send OnIneriorVehicleData notifications to apps

yang1070, 2018-10-09:
### Bug Report
If there is no change on module data, SDL/HMI shall not send OnIneriorVehicleData notifications to apps.

Observed SDL or HMI send OnIneriorVehicleData with empty control data to mobile apps.

### Reproduction Steps
1. launch core and hmi
2. register two RC media app to sdl and activate both from HMI
3. from app1 send GetIneriorVehicleData with moduleType=HMI_SETTINGS and subscribe=true
4. from app2 send GetIneriorVehicleData with moduleType=HMI_SETTINGS and subscribe=true
5. on HMI check the HMI settings: display mode, distance unit and temperature unit
6. from app1 send SetIneriorVehicleData with moduleType=HMI_SETTINGS and module data, make sure that all three hmi settings are included but the desired values are the same as the current values displayed on hmi

### Expected Behavior
both apps shall not receive OnIneriorVehicleData notifications

### Observed Behavior
both app receive OnIneriorVehicleData notifications, with empty 'hmiSettingsControlData`

### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 5.0.0RC2
* HMI version: develop commit f134a7a887675f6f1fa710a28e8612ea78b557e0
* TCP connection from app to core

GetmanetsIrina, 2018-10-09:
This is HMI issue, HMI must not send OnIneriorVehicleData in case not change any data.

JackLivio, 2018-10-15:
Closing, not core issue

closed 2018-10-15 by JackLivio