smartdevicelink/sdl_core #2664: SDL does not cut parameters not related to defined module type from RC.SetInteriorVD

GetmanetsIrina, 2018-10-09:
### Bug Report
SDL does not cut parameters not related to defined module type from RC.SetInteriorVD 

##### Reproduction Steps
1. RC app is registered and activated
2. App requests SetInteriorVD(CLIMATE) with valid parameters for defined module type and data  radioControlData.

##### Expected Behavior
SDL receives SetInteriorVD and cuts all data which is not related to defined module and sends request to HMI with CLIMATE data only.

##### Observed Behavior
SDL resends RC.SetInteriorVD request to HMI with radioControlData

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/5.0.0_RC2
* Testing Against: ATF develop, script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2103.

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

yang1070, 2018-10-11:
@JackLivio @theresalech @lauratonwe 
Issue reproduced, please triage.

jacobkeeler, 2018-10-15:
Closing with the merge of #2675

closed 2018-10-15 by jacobkeeler