smartdevicelink/sdl_core #2670: SDL process GetInteriorVehicleData with resultCode DISALLOWED after IGN_OFF and IGN_ON

GetmanetsIrina, 2018-10-11:
[Delete any non-applicable sections, but we may ask for more information.]

### Bug Report
SDL process GetInteriorVehicleData with resultCode DISALLOWED after IGN_OFF and IGN_ON

##### Reproduction Steps
1. Define module type module_1 and in default section in preloaded_pt.json
2. Update groups in default with "RemoteControl" group
3. Register RC app
4. App requests successful GetInteriorVD with module_1
5. Perform IGN_OFF
6. Perform IGN_ON
7. Register same RC app
8. Request GetInteriorVD with allowed module_1

##### Expected Behavior
SDL process  GetInteriorVD with allowed module_1 successful and resend request to HMI

##### Observed Behavior
SDL responds with DISALLOWED resultCode to GetInteriorVD

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/8da2a76d7dbde7845c1116034f8a60490f4fad0c
* HMI version: https://github.com/smartdevicelink/sdl_hmi/commit/51a4ace4be6bab9613de87d3831522ad443acd68

SDL logs: 
[logs.tar.gz](https://github.com/smartdevicelink/sdl_core/files/2469848/logs.tar.gz)

GetmanetsIrina, 2018-10-11:
@JackLivio @theresalech @lauratonwe 
Issue reproduced, please triage.

LuxoftAKutsan, 2018-10-12:
Reproduced after merging https://github.com/smartdevicelink/sdl_core/pull/2609

jordynmackool, 2018-10-12:
After investigation, this appears to be an issue present in `master` making it not specific to the `release candidate 5.0.0`. It was determined that this bug should only affect the policies of app with default policies where default includes RC permissions making it a lower priority.

jacobkeeler, 2018-10-16:
Closing with the merge of #2676

PR #2676 (2018-10-16): Load module types from referenced group for app_policies
Fixes #2670

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested with script attached to #2670

### Summary
If application have default policies, sqldb does not store module types for this application separately.

During loading db SDL should check that application have default policies
and select from module_types sql table sing  "default" key but not appid key.

In function `GatherApplicationPoliciesSection` already exist check for default policies :

```
    if (IsDefaultPolicy(app_id) [... cut]
closed 2018-10-16 by jacobkeeler