smartdevicelink/sdl_core #2672: SDL lacks a mechanism to invalid cached interior vehicle data

yang1070, 2018-10-11:
### Bug Report
SDL lacks a mechanism to invalid cached interior vehicle data.
Within interior vehicle data set, some data depends on other data. For example, most radio data depends on radioEnabled state. 
(1) If radio is disabled, most radio data in not available. 
(2) If HD radio is disabled, HD radio related data is not available.
(3) If climate control is not enabled in vehicle, most climate data is not available.
SDL has a cache of interior vehicle data set if there is at least an app subscribe to a module. However, sdl does not know the dependencies within the data. And there is no API defined between core and hmi to invalid cached data. 

##### Reproduction Steps
1. register and activate a remote control app.
2. from HMI, switch to radio and make sure it is turned on/ enabled
3. from app, send a `GetInteriorVechileData` with `moduleType=RADIO` and `susbcribe=true`
4. SDL will forward the request to HMI and upon success, SDL will create a data cache for radio data, send all  available data back to the app
5. from HMI, turn off radio. HMI will send `OnInteriorVechileData` to SDL regarding `radioEnable=false`
6. from the app or anther app, send a a `GetInteriorVechileData` with `moduleType=RADIO`

##### Expected Behavior
SDL send available valid radio data back to the app in the response.

##### Observed Behavior
SDL send the cached radio data to the app, even the data is not valid any more.

##### OS & Version Information
OS/Version: Ubuntu 16.04
SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/5.0.0_RC2
test remote control regarding SEAT

### Suggestions
This may not get solved within 5.0 release as we may need a new proposal. 
A potential method is to add a new RPC or add a parameter in OnInteriorVechileData RPC to mark data is not available, or add multi parameters, each corresponding to a single data or a group of data. Once there is a change on HMI / vehicle side, for example turn off radio or turn off climate, HMI shall send such notifications to sdl to invalidate the cached data.
And SDL shall mark cached data not available, and forward GetInteriorVechileData  to HMI.

yang1070, 2018-10-11:
@JackLivio @lauratonwe @theresalech new issue found, please triage.

JackLivio, 2018-10-11:
@yang1070 I think this only applies to `radioEnable` and `hdRadioEnable`. Currently there is no "climateEnable" for the climate module. 

I think the fix for this issue can be implemented without the need for a new proposal or hmi rpc. 

In the construction of the onInteriorVehicleData/getInteriorVehicelDataResponse to the mobile device, if `radioEnable` is false, then we can filter out all of the other radio module parameters. If `hdRadioEnable` is false we can filter out just the `hdChannel` parameter.

I don't think we need to clear them from the cache because when the radio is re-enabled, all of the previous radio states are resumed to previous settings. Also the hmi can resent onInteriorVehicle data cache if it needs to update anything after re-enabling the radio. 

Regarding Climate, I am kind of surprised there is no `climateEnable` param or climateOn/Off button defined in the API, although adding that functionality would require a separate proposal. Without that if the Climate is "disabled" the hmi will have to send all 0/false values.

yang1070, 2018-10-15:
@JackLivio 
- There shall be a separate proposal for adding `climateEnable` parameter.
- In sdl HMI or vehicle radio control unit, there are rules to remove other radio parameters when `radioEnable=false` and hd radio related parameters when `hdRadioEnable=false`. And the result will be reflected in the response of `GetInteriorVehicelData` or `OnInteriorVehicelData` notifications. There is no problem before we have a  InteriorVehicelData Cache. For example, when a user turns of radio, there will    be a single parameter `radioEnable=false`  in the `OnInteriorVehicelData` notification. When the user turns on radio, all available parameters (especially read only RDS and SID data) with updated value are included in the notification. If an app send  `GetInteriorVehicelData` request when the radio is disabled, it can only get `radioEnable=false`  and no other parameters. If an app send  `GetInteriorVehicelData` request when the climate is disabled,  it will get an error `data not available`.
After introduce the data  cache, the `GetInteriorVehicelData` request will not be forwarded to HMI, SDL will include everything in cache in the response. This is a big difference between with and without cache.
The problem is that sdl core does not have these special logic to filter out certain parameters when `radioEnable=false` or `hdRadioEnable=false` or when climate is not enable in the vehicle. Are you suggesting to add these logic in sdl core? That certainly also works.

JackLivio, 2018-10-15:
> The problem is that sdl core does not have these special logic to filter out certain parameters when radioEnable=false or hdRadioEnable=false or when climate is not enable in the vehicle. Are you suggesting to add these logic in sdl core? That certainly also works.

Yes that was the goal of this PR. [https://github.com/smartdevicelink/sdl_core/pull/2677](https://github.com/smartdevicelink/sdl_core/pull/2677)

PR #2677 (2018-10-15): Add radio module filtering if radioEnable or hdRadioEnable are false
Fixes #2672 

This PR is **NOT ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
1. register and activate a remote control app.
2. from HMI, switch to radio and make sure it is turned on/ enabled
3. from app, send a `GetInteriorVechileData` with `moduleType=RADIO` and `susbcribe=true`
4. SDL will forward the request to HMI and upon success, SDL will create a data cache for radio data, send all  available data back to the app
5. from HMI, turn off radio. HMI will send `OnInteriorVechileData` to  [... cut]
closed 2018-10-15 by JackLivio