smartdevicelink/sdl_core #2678: SDL responds with 4.5.0 version in RegisterAppInterface

dboltovskyi, 2018-10-15:
### Bug Report
SDL responds with `4.5.0` version in RegisterAppInterface however in Mobiler API `5.0.0` is defined

##### Reproduction Steps
1. Register Mobile App
2. Check RegisterAppInterface response

##### Expected Behavior
SDL responds with `5.0.0` version that is correspond to current Mobile API

##### Observed Behavior
SDL responds with `4.5.0` version that is correspond to current Mobile API

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 7b41964
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
* Script:  [MobileVersioning/VersionNegotiation/001_register_legacy_app](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/SDL5_0/MobileVersioning/VersionNegotiation/001_register_legacy_app.lua)
* ATF: [develop](https://github.com/smartdevicelink/sdl_atf/tree/develop)
* Logs: [logs.zip](https://github.com/smartdevicelink/sdl_core/files/2479340/logs.zip)

AStasiuk, 2018-10-15:
@theresalech @JackLivio, @jacobkeeler, @yang1070. Please triage. Thanks

JackLivio, 2018-10-15:
@AStasiuk @dboltovskyi I am not seeing this issue. I ran the test script 001_register_legacy_app.lua and it passes.

```
===== RegisterAppInterface_Legacy_App_Case : 
MOB->SDL  (15 10 2018 14:05:42, 657) [rpcFunction: RegisterAppInterface, sessionId: 1, version: 2, frameType: 1, encryption: false, serviceType: 7, frameInfo: 0, messageId: 5, binaryDataSize: 0] : {"syncMsgVersion":{"majorVersion":2,"minorVersion":2},"ngnMediaScreenAppName":"SPT","appHMIType":["DEFAULT"],"isMediaApplication":true,"appName":"SyncProxyTester","languageDesired":"EN-US","deviceInfo":{"firmwareRev":"firmwareRev","os":"os","osVersion":"osVersion","maxNumberRFCOMMPorts":5,"carrier":"carrier","hardware":"hardware"},"appID":"123456","vrSynonyms":["VRSyncProxyTester"],"ttsName":[{"type":"TEXT","text":"SyncProxyTester"}],"hmiDisplayLanguageDesired":"EN-US"} 
SDL->MOB (15 10 2018 14:05:42, 780) [rpcFunction: RegisterAppInterface, sessionId: 1, version: 2, frameType: 3, encryption: false, serviceType: 7, frameInfo: 0, messageId: 9, binaryDataSize: 0] : {"syncMsgVersion":{"majorVersion":4,"minorVersion":5,"patchVersion":1}
```

closed 2018-10-15 by JackLivio

JackLivio, 2018-10-15:
Using ATf test scripts commit: bb6dcae7103a3dfae3bc946ae31043d642a15172
Using Core commit: 96aa5668f36e96c2cd5d93b2b29a4e2d601f8426

JackLivio, 2018-10-15:
```
==============================
Start 'test_scripts/SDL5_0/MobileVersioning/VersionNegotiation/001_register_legacy_app.lua'
==============================
=====================================================================================================
Description: ATF test script
Severity: Major
=====================================================================================================
--- Preconditions -----------------------------------------------------------------------------------
[14:19:23,858] Clean_environment                                                                     [SUCCESS] (2 ms)
SDL pid 94737
tcp        0      0 127.0.0.1:8087          0.0.0.0:*               LISTEN     
HMI initialized
HMI is ready
Mobile connected
[14:19:25,514] Start_SDL__HMI__connect_Mobile__start_Session                                         [SUCCESS] (862 ms)
[14:19:26,412] RAI                                                                                   [SUCCESS] (373 ms)
[14:19:26,799] Activate_App                                                                          [SUCCESS] (538 ms)
[14:19:27,353] UnregisterAppInterface_Positive_Case                                                  [SUCCESS] (76 ms)
--- Test --------------------------------------------------------------------------------------------
[14:19:27,436] RegisterAppInterface_Legacy_App_Case                                                  [SUCCESS] (296 ms)
--- Postconditions ----------------------------------------------------------------------------------
[14:19:27,748] Stop_SDL                                                                              [SUCCESS] (2 ms)
Total executing time is 6s 64ms (summary 6064ms)
==============================
Finish 'test_scripts/SDL5_0/MobileVersioning/VersionNegotiation/001_register_legacy_app.lua'
==============================
jack@ubuntu:~/Projects/sdl_atf$ 

```

dboltovskyi, 2018-10-15:
@JackLivio On a commit 1ef06f5 script is passed. Thank you for update.