smartdevicelink/sdl_core #2683: SDL processes Light name values from the default capabilities incorrectly

LuxoftAKutsan, 2018-10-16:
[Delete any non-applicable sections, but we may ask for more information.]

### Bug Report

SDL must use the default values in case HMI does not provide values.
in hmi_capabilities.json :
```
          "lightControlCapabilities": {
                        "moduleName": "light",
                        "supportedLights":[
                              {
                                 "statusAvailable":true,
                                 "densityAvailable":true,
                                 "name":"FRONT_LEFT_HIGH_BEAM",
                                 "rgbColorSpaceAvailable":true
                              },
                              ....
                        ]
         }
```
the name for each light must be converted from a string to an enum value in accordance with HMI_API.xml

##### Reproduction Steps
1. Send from HMI not valid capabilities. SDL should load capabilities from hmi_capabilities.json  
2. Send SetInteriorVehicleData with Module type LIGHT and light any name.

##### Expected Behavior
SDL process SetInteriorVehicleData according hmi_capabilities.json 

##### Observed Behavior
SDL respond with `UNSUPPORTED_RESOURCE` + "There is no such light name in capabilities"

##### OS & Version Information
* OS/Version: Ubuntu 16.04 x86_64
* SDL Core Version: release/5.0.0 17c3f5114533f954d9afea8166c1d3d7b3adff2a
* Testing Against: Automated script :exclamation: **TODO: Attach script**

LuxoftAKutsan, 2018-10-16:
Have taken it in work

AStasiuk, 2018-10-16:
@theresalech, @lauratonwe, @JackLivio please triage.

theresalech, 2018-10-16:
@AStasiuk please provide the automated script referenced in the `OS & Version Information` section of the issue.  We will need this in order to review the issue and PR.

AStasiuk, 2018-10-16:
@theresalech script will be provided soon

theresalech, 2018-10-16:
It's been determined that the root cause of this issue exists in master, and was not something introduced by the 5.0 Release, so we are not able to include this fix in the 5.0 Release.  This issue can be prioritized for the next release of SDL Core if the SDLC Steering Committee agrees.

jacobkeeler, 2019-03-04:
Closing with the merge of #2684

PR #2684 (2019-03-04): Fix the Light Name conversion from string to enum value in default capabilities
Fixes #2683

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Automated script : https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2118

### SummarySDL must use the default values in case HMI does not provide values.
in hmi_capabilities.json :

          "lightControlCapabilities": {
                        "moduleName": "light",
                        "supportedLights":[
                              {
                                 "statusAvailable":true [... cut]
closed 2019-03-04 by jacobkeeler