smartdevicelink/sdl_core #2685: SDL does not provide the list of groups with user_consent_prompt to HMI

GetmanetsIrina, 2018-10-16:
### Bug Report
SDL does not provide the list of groups with user_consent_prompt to HMI and it is not possible to change permissions for this groups( allow or disallow)

##### Precondition
SDL is built with EXTERNAL_PROPRIETARY

##### Occurrences
Sometimes

##### Reproduction Steps
1. Register app and activate it
2. Provide device consent
3. Perform PTU with groups with user_consent_prompt
4. Go to Settings->App permission->"Get list of permissions"

##### Expected Behavior
SDL responds GetListOfPermissions with names and ids of groups in allowedFunctions array

##### Observed Behavior
SDL responds with empty allowedFunctions array

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/17c3f5114533f954d9afea8166c1d3d7b3adff2a
* Testing Against: Mobile 20180912
File for update:
[ptu_5.0.0_consent.json.tar.gz](https://github.com/smartdevicelink/sdl_core/files/2484910/ptu_5.0.0_consent.json.tar.gz)

AStasiuk, 2018-10-16:
@theresalech, @lauratonwe, @JackLivio please triage.

JackLivio, 2018-10-16:
@AStasiuk @GetmanetsIrina We attempted this multiple times on multiple machines and were unable to replicate the issue.

GetmanetsIrina, 2018-10-17:
@theresalech , @AStasiuk , @JackLivio , @lauratonwe 
This is local environment issue. 
Issue is not reproduced(20 attempts was performed) after environment refresh.

closed 2018-10-17 by GetmanetsIrina