smartdevicelink/sdl_core #2686: Remove duplicate policy_handler_observer.h

HSavynetska, 2018-10-17:
### Bug Report
Remove duplicate policy_handler_observer.h 

##### Steps
1. The path to the file "policy_handler_observer.h":
src/components/application manager/include/application manager/policies/regular/policy_handler_observer.h
2. Make sure that duplicate policy_handler_observer.h is created.

##### Expected Behavior
Duplicate policy_handler_observer.h is not created.
**Note**: Make sure that removing this file does not cause problem for other configurations.

##### Observed Behavior
Duplicate policy_handler_observer.h is created.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

JackLivio, 2018-10-17:
@HSavynetska Please add version information. Please do not mark every issue as high priority.

iCollin, 2019-10-18:
Thanks for reporting!
I do not see a duplicate `policy_handler_observer.h` file.
I am assuming this was fixed at some point in the last year.

closed 2019-10-18 by iCollin