smartdevicelink/sdl_core #2693: Prevent creation of SSL context during certificate updating

HSavynetska, 2018-10-17:
### Bug Report
 Prevent creation of SSL context during certificate updating 

#### Steps
1. Need added lock for CryptoManagerImpl

##### Expected Behavior
 Prevent creation of SSL context during certificate updating 

##### Observed Behavior
Sometimes SDL trigger redundant update, fix by adding lock for CryptoManagerImpl for preventing race conditions. One thread is trying to update " OnCertificateUpdated" and another trying to "CreateSSLContext".

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

HSavynetska, 2018-10-23:
Updated issue according to requirements: added versions, detailed description

LitvinenkoIra, 2019-12-13:
Trying to reproduce the issue

iCollin, 2020-02-05:
@LitvinenkoIra did you have success reproducing this issue? It seems to me the script is passing on develop and failing on the fix branch.

ydementieiev, 2020-02-10:
@iCollin you are right, this bug not reproduction using attached script, we tried to do this with unit test but finished without good result, having studied the problem, we believe that an attached pull request is a potential fix. I wrote why in the [Summary to PR](https://github.com/smartdevicelink/sdl_core/pull/2777). Thank you

iCollin, 2020-02-11:
Closed via #2777

closed 2020-02-11 by iCollin