smartdevicelink/sdl_core #2697: Fix moduleType save to the policy database

HSavynetska, 2018-10-17:
### Bug Report
Fix moduleType save to the policy database 

##### Steps
1.Need added missing logic of saving this parameter
into the database.

##### Expected Behavior
Fixed moduleType save to the policy database 

##### Observed Behavior
There was a missed logic of saving  of moduleType parameter
of application into the database. SDL does not save this
parameter into the database when application get default
policy permissions. As a result, on the next ignition
circle SDL loads nothing from database for this application
as nothing was saved. As a result SDL rejects all RC related
requests due to wrongly set moduleType parameter.

##### OS & Version Information
* OS/Version: 
* SDL Core Version: 
* Testing Against:

jacobkeeler, 2018-10-17:
Duplicate of https://github.com/smartdevicelink/sdl_core/issues/2670

closed 2018-10-17 by jacobkeeler

hhussei7, 2018-10-24:
@jacobkeeler This PR should be re-opened and looked at.  If we follow the link to #2670 then to #2676 and look at the changes we don't see the changes in the PR. 
We see that the PR https://github.com/smartdevicelink/sdl_core/pull/2618 is still open but the issue is closed.