smartdevicelink/sdl_core #2716: sdl does not include binary data in the OnSystemRequest when requestType=OEM_SPECIFIC

yang1070, 2018-10-24:
### Bug Report
sdl does not include binary data in the OnSystemRequest when requestType=OEM_SPECIFIC

##### Reproduction Steps
- launch core and hmi
- register a RC media app to sdl and activate both from HMI
- prepare a valid json file, for example with contents {"key":"value"}
- on HMI press the button "System Request"
- a pop up is shown
- select "system request reason" as OME_SPECIFIC
- type anything in the "requestSubType", for example "testSubType"
- type  the correct path and file name of the prepared json file
- press send onSystemRequest button
- app receives the onSystemRequest notification

##### Expected Behavior
onSystemRequest message shall include the bulkdata, with contents  {"key":"value"}

##### Observed Behavior
bulkdata is null

##### OS & Version Information
OS/Version: Ubuntu 16.04
SDL Core Version: 5.0.0
HMI version: develop commit f134a7a887675f6f1fa710a28e8612ea78b557e0

yang1070, 2018-10-24:
in file https://github.com/smartdevicelink/sdl_core/blob/d36316738785c96dab2ee892762ed08c059fffde/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/mobile/on_system_request_notification.cc

line 109
 if (mobile_apis::RequestType::PROPRIETARY == request_type) {

shall be 
 if (mobile_apis::RequestType::PROPRIETARY == request_type || mobile_apis::RequestType::OEM_SPECIFIC==request_type) {

AStasiuk, 2018-10-30:
Analyzing this issue. 
Working on providing PR with fix for this issue.
Assigned to @AByzhynar according to  https://d83tozu1c8tt6.cloudfront.net/media/resources/SDL_GitHub_BestPractices.pdf

AByzhynar, 2018-10-30:
@yang1070 You are absolutely right with the place in SDL where it should be changed but I can not find information in related proposal that binary data should be added into `OEM_SPECIFIC` Request Type. 
Here is the proposal: [Expandable Design](https://github.com/AByzhynar/sdl_evolution/blob/master/proposals/0083-Expandable-design-for-proprietary-data-exchange.md)
Could you be so kind and point this expected behavior in proposal above. Thank you.

yang1070, 2018-10-30:
The proposal is for `exchange data` between mobile and vehicle, without bulk data included in the `OnSystemRequest`, it is not possible to send any   `OEM_SPECIFIC` data from vehicle to mobile.

iCollin, 2019-08-13:
fixed in #2756

PR #2734 (2018-11-19): Add binary data to OEM_SPECIFIC request type in OnSystemRequest notification
Fixes #2716 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF script is created in PR: [2132](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2132)

### Summary
Root cause of the issue is in missed logic of handling `OnSystemRequest` notification sent from HMI with `OEM_SPECIFIC` request type and binary data.
Fix adds binary data to `OEM_SPECIFIC` request type in `OnSystemRequest` notification being sent to mobile application.

### CLA
- [x] I have signed [... cut]
closed 2019-08-13 by iCollin