smartdevicelink/sdl_core #2721: SDL sends EndService for Audio/Video twice in case USER_EXIT

dboltovskyi, 2018-10-26:
### Bug Report
SDL sends EndService request to Mobile App for Audio/Video service twice in case user exits from app on HMI

##### Occurrence
Sometimes (10-20%)

##### Reproduction Steps
1. Register streaming App (NAVI or PROJECTION)
2. Activate App
3. Start Audio or Video streaming
4. Perform User Exit by sending `BC.OnExitApplication(USER_EXIT)` from HMI

##### Expected Behavior
SDL sends EndService request only once

##### Observed Behavior
SDL sends EndService request twice
`Navigation.StopStream` request to HMI is also sent twice

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [develop](https://github.com/smartdevicelink/sdl_core/commit/d36316738785c96dab2ee892762ed08c059fffde)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
* [MobileProjection/Phase1/015_Restore_video_streaming_from_NONE](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase1/015_Restore_video_streaming_from_NONE.lua)
* [MobileProjection/Phase1/016_Restore_audio_streaming_from_NONE](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase1/016_Restore_audio_streaming_from_NONE.lua)

##### Logs
[logs.tar.gz](https://github.com/smartdevicelink/sdl_core/files/2518464/logs.tar.gz)

E-SAITO-TMC, 2020-04-20:
@theresalech -san This issue is a medium priority for Toyota. I would like know if this issue occur in the latest version. And could you please include this issue in your plan? Thank you.

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@dboltovskyi, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-04-22:
@theresalech Please notice issue is not reproduced at https://github.com/smartdevicelink/sdl_core/tree/release/6.1.0 (a85ef58)
So it can be closed.

closed 2020-04-22 by theresalech