smartdevicelink/sdl_core #2732: Crash observed on extra dlClose in LoadPlugin function

yang1070, 2018-10-30:
### Bug Report
We are loading a plugin.  So, after dlopen() it becomes a block in memory.   All variables initialized and all functions available. Then we create the instance of plugin.  The Application should be able to use the plugin APIs of created instance, which references the plugin.
However once we added dlclose() we unloaded all objects referenced by the handle. 

From: http://pubs.opengroup.org/onlinepubs/7908799/xsh/dlclose.html:
dlclose() is used to inform the system that the object referenced by a handle returned from a previous dlopen() invocation is no longer needed by the application.

From: http://www.qnx.com/developers/docs/6.5.0/index.jsp?topic=%2Fcom.qnx.doc.neutrino_lib_ref%2Fd%2Fdlclose.html
The dlclose() function disassociates a shared object opened by dlopen() from the calling process. An object's symbols are no longer available after it's been closed with dlclose(). All objects loaded as a result of the closed objects dependencies are also closed

LoadPlugin function returns RPCPluginPtr(plugin) this is the instance of the class RCRPCPlugin, this instance is expected to use outside of this function (LoadPlugin).

This change https://github.com/smartdevicelink/sdl_core/commit/35defc42777c57412cbf8b793489ab0dac502f4e must be reverted.

##### Reproduction Steps
launch sdl core with plugins

##### Expected Behavior
no crash

##### Observed Behavior
crash

##### OS & Version Information
* OS/Version: Linux and QNX
* SDL Core Version: 5.0.0

yang1070, 2018-10-30:
@lauratonwe @JackLivio Please triage.

lauratonwe, 2018-10-30:
IVI team is integrating 5.0 with Sync4 and is experiencing this issue.  They have been able to reproduce it both on Linux and QNX.  They consider it a very high priority.

JackLivio, 2018-11-19:
Reverted commit. Fixed in 5.0.1 #2756

closed 2018-11-19 by JackLivio