smartdevicelink/sdl_core #2735: EmergencyEvent->maximumChangeVelocity data type is incorrect in HMI API

atiwari9, 2018-10-31:
### Bug Report
_maximumChangeVelocity_ param for _EmergencyEvent_ data type is mismatched between HMI and Mobile API. HMI API has data type _VehicleDataEventStatus_(**Incorrect**) and Mobile API has data type _Integer_ (**Correct**) as _maximumChangeVelocity_ signal ranges from 0-255

HMI API (**Incorrect**):
```
<param name="maximumChangeVelocity" type="Common.VehicleDataEventStatus" mandatory="true">
    <description>References signal "VedsMaxDeltaV_D_Ltchd". See VehicleDataEventStatus.</description>
</param>
```
Ref: https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/HMI_API.xml#L3134

Mobile API(**Correct**):
```
<param name="maximumChangeVelocity" type="Integer" minvalue="0" maxvalue="255" mandatory="true">
    <description>References signal "VedsMaxDeltaV_D_Ltchd". Change in velocity in KPH.  Additional reserved values:
        0x00 No event
        0xFE Not supported
        0xFF Fault
    </description>
</param>
```
Ref: https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/interfaces/MOBILE_API.xml#L1772

**Expected:** Data type for _maximumChangeVelocity_ in HMI API needs to be corrected so that it matches with Mobile API

atiwari9, 2019-01-17:
@JackLivio  - Can you please Triage this issue, data type for MaxChangeVelocity is incorrect in HMI API

atiwari9, 2019-03-05:
@JackLivio  @jacobkeeler  - What was decided for this issue?

jordynmackool, 2019-09-16:
@atiwari9 Do you plan on submitting a PR fix for this issue?

atiwari9, 2019-09-16:
@jordynmackool - https://github.com/smartdevicelink/sdl_evolution/pull/820

theresalech, 2019-09-17:
@atiwari9 an Evolution proposal is not needed for this issue.  The Livio team will plan to submit an sdl_core PR to resolve this issue within the 6.0 release.

jacobkeeler, 2019-09-23:
Closing with the merge of #3040

PR #3040 (2019-09-23): Fix definition of maximumChangeVelocity in HMI API

Fixes #2735 

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Summary
Fix definition of maximumChangeVelocity in HMI API

### Changelog
##### Breaking Changes
* Data type of maximumChangeVelocity in HMI API was changed

##### Bug Fixes
* Fix definition of maximumChangeVelocity in HMI API

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-09-23 by jacobkeeler