smartdevicelink/sdl_core #2763: SDL 0204- Support running the same app from multiple devices at the same time

jordynmackool, 2018-11-21:
Proposal: [Support running the same app from multiple devices at the same time](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0200-Removing-URL-Param-Max-Length.md)

> This proposal proposes that SDL shall support running the same app from multiple devices at the same time. Currently, SDL only allows one instance of the same app to register with SDL. However, there are valid use cases for running the same app from two or more devices at the same time.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/618

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions. The revisions will include updating the descriptions of ttName and vrSynonyms as discussed in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/618#issuecomment-440345476).
In addition, the author will address numbers 2 and 4 in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/618#issuecomment-440308172).
•	2. I would specify in proposal that after policy update, it will be applied for both of registered applications
•	4. Application on different device should share same usage statistics, because they use same policy app_id. this should be added to the proposal.

jordynmackool, 2018-11-21:
Proposal has not yet been update, Once it is updated I will create an issue in the core repo.

AStasiuk, 2018-11-23:
Hi @jordynmackool,
the link in the description  "Support running the same app from multiple devices at the same time" led to another proposal:
https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0200-Removing-URL-Param-Max-Length.md
and not to https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0204-same-app-from-multiple-devices.md

jordynmackool, 2018-11-23:
Hi @AStasiuk, thanks for catching that. I have updated the link.

theresalech, 2019-08-19:
Closed via #2960

PR #2960 (2019-08-16): Support running the same app from multiple devices at the same time
Fixes #2763 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes. Just updates for some descriptions

### Testing Plan
* Unit tests were updated/extended
* ATF scripts 

### Summary
Added possibility for registration and usage of the same application from multiple devices.

### Changelog
* Updated interfaces.
* Extended registration flow
* Added usage of device id for both policy modes: proprietary/external proprietary
* Unit tests were fixed according to the code changes

##### Bug Fixes
* Fixed sendi [... cut]
closed 2019-08-19 by theresalech