smartdevicelink/sdl_core #2785: Possible buffer over-run when parsing corrupted BSON data

shoamano83, 2018-12-13:
### Bug Report
SDL Core uses BSON library's `bson_object_from_bytes()` function when parsing BSON data received from mobile libraries. Unfortunately, it does not take into account the size of input data. If the input data is corrupted or malformed, the function may continue reading beyond the allocated buffer and may cause undefined behavior.

Please refer to BSON library's issue: https://github.com/smartdevicelink/bson_c_lib/issues/21

##### Reproduction Steps
Please refer the repro steps of BSON library' issue page. Unit tests are provided in the library.

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: Latest `develop` branch should be affected
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
N/A

shoamano83, 2018-12-18:
Related issue: https://github.com/smartdevicelink/sdl_android/issues/943

Android library (4.7.2) generates an End Service frame with 4-byte raw hash ID instead of BSON document. SDL Core receives it and tries to parse the bytes as BSON document, so it will read data beyond the buffer boundary.

jacobkeeler, 2019-05-29:
Closing with the merge of #2908 and #2786

closed 2019-05-29 by jacobkeeler