smartdevicelink/sdl_core #2787: Rare SDL Core crash upon termination

shoamano83, 2018-12-13:
### Bug Report
When terminating SDL Core, a crash with "double free or corruption" may occur.

##### Reproduction Steps
1. Check out latest `develop` branch: commit aadea479ba2ac48bee55699265559c14348b2274
2. Give a build
3. Run `./start.sh`
4. After several seconds, terminate with Ctrl-C

##### Expected Behavior
SDL Core terminates without any errors.

##### Observed Behavior
A memory error with "double free or corruption" is observed. See attached logs.
Repro ratio is once out of 23 times.

##### OS & Version Information
* OS/Version: Ubuntu 16.04 LTS (64-bit)
* SDL Core Version: commit aadea479ba2ac48bee55699265559c14348b2274 from `develop`
* Testing Against: N/A

##### Test Case, Sample Code, and / or Example App
[sdlcore_memory_issue_console.txt](https://github.com/smartdevicelink/sdl_core/files/2676010/sdlcore_memory_issue_console.txt)
[sdlcore_memory_issue_gdb.txt](https://github.com/smartdevicelink/sdl_core/files/2676011/sdlcore_memory_issue_gdb.txt)
[log_files.tar.gz](https://github.com/smartdevicelink/sdl_core/files/2676018/log_files.tar.gz)

shoamano83, 2018-12-13:
When I run the executable with valgrind, I see a lot of memory errors during exit handler. I guess static variables are deleted multiple times and it may cause a crash. Please refer to attached file for valgrind's output: [valgrind_output.txt](https://github.com/smartdevicelink/sdl_core/files/2676045/valgrind_output.txt)

iCollin, 2019-10-18:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2939

closed 2019-10-18 by iCollin