smartdevicelink/sdl_core #2791: [SDL 0211]- Service Status Update to HMI

jordynmackool, 2018-12-20:
Proposal: [Service Status Update to HMI](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0211-ServiceStatusUpdateToHMI.md)

> This proposal is about adding a new RPC called 'OnServiceUpdate' which will be used by the HMI layer to know the status of certain services from SDL Core.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/641#issuecomment-448337280

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions on 2018-12-18. The author is to update the proposal to include both the Audio and RPC services discussed in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/641#issuecomment-447112795) and adding an optional 'appID' parameter to the OnServiceUpdate RPC as noted in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/641#issuecomment-448337280). 

Revisions were made on 2018-12-20

theresalech, 2019-06-12:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0211 Revisions - Service Status Update to HMI](https://github.com/smartdevicelink/sdl_evolution/issues/756).  Implementation should now include `ServiceStatusUpdateReason` instead of `ServiceUpdateReason` and also include `PROTECTION_ENFORCED` and `PROTECTION_DISABLED` as reason codes/elements in the enum.

jacobkeeler, 2019-08-29:
Closing with the merge of #2921

closed 2019-08-29 by jacobkeeler