smartdevicelink/sdl_core #2793: [SDL 0213]- Remote Control - Radio and Climate Parameter Update

jordynmackool, 2019-01-02:
Proposal: [Remote Control - Radio and Climate Parameter Update](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0213-rc-radio-climate-parameter-update.md)

> This proposal changes the minimum index of HD radio sub-channels from 1 to 0.
>In addition, we propose to add a new parameter climateEnable to ClimateControlData which will allow an application to power climate control on or off.

https://github.com/smartdevicelink/sdl_evolution/issues/643

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions. The author is to update this proposal to state that `availableHDs` and `availableHdsAvailable` are deprecated entirely if the functionality is subsumed by `availableHDChannels` as finalized in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/643#issuecomment-448363428). 

Revisions were made on 2019-01-02

FStoikov, 2019-02-15:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-2691

theresalech, 2019-06-19:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0213 Revisions - rc-radio-climate-parameter-update](https://github.com/smartdevicelink/sdl_evolution/issues/761).  Implementation should now include the following:
- Change deprecated availableHDs  minvalue back to 1. The API should not change the rules for a parameter in the same version that the parameter is being deprecated in.
- Add correct history line to availableHDs.
- Note the `hdChannel` change incurs a major version change.
- Update `since` values to 5.x and 6.x to highlight minor/major api changes.
- Give history tag to `availableHDsAvailable` for pre-deprecated version.

Note that this is now a breaking change to Core.

Please reference the issue and proposal markdown file for more details.

theresalech, 2019-06-24:
Closed via #2920

PR #2920 (2019-06-21): Feature/Remote Control - Radio and Climate Parameter Update
Fixes #2793

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2191
### Summary
This proposal changes the minimum index of HD radio sub-channels from 1 to 0.
In addition, we propose to add a new parameter climateEnable to ClimateControlData which will allow an application to power climate control on or off.

##### Enhancements
 * Refactor CutOffReadOnlyParams() method for better readable.
 * Update SetInteriorData Unit tests be [... cut]
closed 2019-06-24 by theresalech