smartdevicelink/sdl_core #2798: SDL crashes while registering app.

Nilesh-harman, 2019-01-17:
SDL crashes while registering app.
[crash_GetAppIdForSending.txt](https://github.com/smartdevicelink/sdl_core/files/2768685/crash_GetAppIdForSending.txt)
[crash_GetCurrentDeviceId.txt](https://github.com/smartdevicelink/sdl_core/files/2768687/crash_GetCurrentDeviceId.txt)

### Bug Report
SDL crashed two times(attached the backtrace).
1.GetAppIdForSending
2.GetCurrentDeviceId
Please refer below for the code flow found in the back trace. I have attached the backtrace as well.
1)RegisterAppInterfaceRequest::Run()->  RegisterAppInterfaceRequest::SendRegisterAppInterfaceResponseToMobile()
--->PolicyHandler::OnAppRegisteredOnMobile()--->PolicyManagerImpl::OnAppRegisteredOnMobile----->PolicyManagerImpl::StartPTExchange()--->PolicyHandler::CanUpdate()
---->PolicyHandler::GetAppIdForSending() --->operator()(const ApplicationSharedPtr app)
 return level_ == app->hmi_level() ? true : false;

2)RegisterAppInterfaceRequest::Run()---> RegisterAppInterfaceRequest::SendRegisterAppInterfaceResponseToMobile()
--->PolicyHandler::OnAppRegisteredOnMobile()--->PolicyManagerImpl::OnAppRegisteredOnMobile()--->PolicyManagerImpl::SendNotificationOnPermissionsUpdated()
--->PolicyManagerImpl::GetCurrentDeviceId()

##### Reproduction Steps
1. Connect iPhone and HU via USB
2. wait for app registration.
3. Only four apps got registered
Note: the iPhone have more than 10 apps. In successful case, all apps get registered.

##### Expected Behavior
All app should register via USB

##### Observed Behavior
Only four apps got registered and SDL crashed

##### OS & Version Information
* OS/Version: IOS
* SDL Core Version: 4.2.1

Please let me know if you need any info.

Thanks,
Nilesh

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
Hi @Nilesh-harman, can you please advise if you are able to reproduce this issue in the latest Core release (6.1.1)?

LuxoftAKutsan, 2020-05-26:
Issue is relate to ios device so cant be reproducible on open sdl. 
Code with crashes still exist in codebase, need deeper investigation to find root cause.

JackLivio, 2020-10-06:
Fixed by #3528

closed 2020-10-06 by JackLivio