smartdevicelink/sdl_core #2803: Resumption doesn't work in the next ignition cycle after Low voltage

LitvinenkoIra, 2019-01-24:
### Bug Report
Resumption doesn't work in the next ignition cycle after Low voltage

### Preconditions:
SDL is up and running - 1st ignition cycle

### Steps:
1. Register non-media App
2. Send LOW_VOLTAGE signal
3. Send IGNITION_OFF signal
4. Start SDL and wait 30 sec (for resuming app from prev ign cycle)
5. Register 2 App
6. Activate 2 app
7. Disconnect transport (unregister 2 app)
8. Connect transport again

##### Expected Behavior
App should be activated automatically

##### Observed Behavior
SDL doesn't send ActivateApp (for the 2 app) to the Applink **(HMI level resumption not working in the current ignition cycle)**

##### OS & Version Information
* OS/Version: Ubuntu 16.04 
* SDL Core Version: 5.0.0
* Testing Against: ATF test script (in the archive) + ATF and SDL logs
[Resumption_after_Low_Voltage.zip](https://github.com/smartdevicelink/sdl_core/files/2791886/Resumption_after_Low_Voltage.zip)

dboltovskyi, 2020-09-16:
Please notice issue is still actual for current `sdl_core` [develop](https://github.com/smartdevicelink/sdl_core/tree/develop) (a36ea82)
Report and logs: [2020-09-16_11-48-43.176.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5233328/2020-09-16_11-48-43.176.tar.gz)

dboltovskyi, 2020-12-15:
Please notice issue is not related to low voltage functionality and can be reproduced without it.
Summary and Description were updated accordingly.
Corresponding new steps to reproduce are implemented in ATF script in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2496

jacobkeeler, 2021-01-14:
Closing with the merge of #3608

PR #3608 (2021-01-14): Fix resumption after unexpected disconnect in current ignition cycle
Fixes #2803 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test script

### Summary
Fix HMI level resumption of application after unexpected disconnect in current ignition cycle by adding corresponding check.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-01-14 by jacobkeeler