smartdevicelink/sdl_core #2805: SDL 0167 - App Services - Weather Service

jordynmackool, 2019-01-24:
Proposal: [SDL 0167 - App Services - Weather Service](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0167-app-services.md#weather)

> SmartDeviceLink has always been limited to a 1 - 1 relationship between module and smartphone apps. However, to create truly compelling experiences this should be expanded. This proposal will aim to introduce a paradigm that allows apps to advertise a "service" they offer that the module and other apps can leverage. This allows users to augment their vehicle's IVI system with off-board solutions. This proposal also details how SDL apps can communicate with each other through the module.

https://github.com/smartdevicelink/sdl_evolution/issues/581

Steering Committee Decision:

> The Steering Committee voted to accept the Weather Service type section of the larger App Services proposal, with the revisions included in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/581#issuecomment-412519648).

The accepted revisions were made noted in this comment on 2019-01-24

jordynmackool, 2019-03-15:
Implementation revision as of 2019.03.15:

`param name="weatherIconImageName" type="String" mandatory="false"/`
 to
`param name="weatherIcon" type="Image" mandatory="false"/`

jacobkeeler, 2019-04-17:
Closing with the merge of #2853

closed 2019-04-17 by jacobkeeler