smartdevicelink/sdl_core #2808: [SDL] Crash seen in SDL upon receiving rpc - OnEmergencyEvent with "enabled:true"

kekapare, 2019-02-07:
### Bug Report
When the app is getting registered, if SDL receives rpc OnEmergencyEvent with "enabled:true", then there is assert failure in UsageStatistics::RecordHmiStateChanged(HMILevel::eType new_hmi_level) due to hmi_level -1 i.e. INVALID_ENUM

##### Reproduction Steps
1. Sygic app is launched on the phone.
2. Connect the Android phone to HU over USB. 
3. At the same time SDL receives rpc  - OnEmergencyEvent with "enabled:true" from HMI.

######Note
Make sure that OnEmergencyEvent with enabled:true is received by SDL at the time when app is getting registered.

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashes with DCHECK failure 

##### OS & Version Information
* OS/Version: Android 8.0.0
* SDL Core Version: 4.2

##### Test Case, Sample Code, and / or Example App
[SDLBacktrace.txt](https://github.com/smartdevicelink/sdl_core/files/2840500/SDLBacktrace.txt)
[SDLLogs.txt](https://github.com/smartdevicelink/sdl_core/files/2840501/SDLLogs.txt)

Please let us know if more information is required.

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
Hi @kekapare, can you please advise if you are able to reproduce this issue in the latest Core release?

kekapare, 2020-05-21:
@theresalech  : RIght now, we are still working on older release for our customer programs. Hence not tried on the latest releases.
Thank you,
Kedar

theresalech, 2020-05-21:
@kekapare, thank you for the quick response!  

@E-SAITO-TMC, please advise if you are able to reproduce this issue in the latest Core Release (6.1.1), and provide any logs and additional information that will assist in determining a resolution.

JackLivio, 2020-10-26:
Fixed by #3546

closed 2020-10-26 by JackLivio