smartdevicelink/sdl_core #2810: Cannot stop Audio stream from the Application, streaming is still sent by SDL to HMI even after the Application is Unregistered

LitvinenkoIra, 2019-02-12:
### Bug Report

The issue was observed when some application is streaming video or audio and unexpected disconnect happened then SDL does not close active service properly and does not send notification to HMI in this case. As a result, HMI is still playing received audio/video data from SDL.

##### Reproduction Steps
Perpoduced on SYNC4 HU.
How To Reproduce:

- 1.Connect an iOS device to SYNC via USB
- 2.Launch Nav App on Mobile device and then on SYNC
- 3.Go to Menu options on the mobile device --> AV Streaming -->Under Audio Streaming Encrypt Video to NONE --> Start session --> Start Audio
- 4. Stop Audio stream 
- 5.Unplug the device from SYNC

##### Expected Behavior
Audio should stop streaming when stopped and stop completely when the application is unregistered

##### Observed Behavior
Cannot stop Audio and the audio is still heard even after the Application is unregistered 

##### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a medium priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if this issue is reproducible in Core 6.1.1?

LuxoftAKutsan, 2020-05-26:
Appropriate sync4 changes not present in open sdl_core. 
Script will be created and changes will be ported

theresalech, 2020-05-26:
Hi @LuxoftAKutsan, thank you! Please let us know once the PR is ready for Livio review.

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

dboltovskyi, 2021-09-17:
@theresalech @E-SAITO-TMC Please notice we unable to reproduce this issue on a current  [release/8.0](https://github.com/smartdevicelink/sdl_core/tree/release/8.0.0) (de58a41) branch.
Comparing `8.0.0` HMI logs against `6.0.1` ([HMI_log_8.0.0.log](https://github.com/smartdevicelink/sdl_core/files/7187865/HMI_log_8.0.0.log) vs [HMI_log_6.0.1.log](https://github.com/smartdevicelink/sdl_core/files/7187868/HMI_log_6.0.1.log)) we can see SDL now sends both `Navigation.OnAudioDataStreaming(false)` and `Navigation.StopAudioStream` messages to HMI.

Could you please check whether this issue reproducible on your end?

closed 2021-10-05 by iCollin