smartdevicelink/sdl_core #2816: SDL 0207 - RPC message protection

jordynmackool, 2019-02-19:
Proposal: [SDL 0207 - RPC message protection](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0207-rpc-message-protection.md)

> This proposal proposes a design to protect (enable encryption for) RPC messages transmitted between a mobile application and the SDL.

https://github.com/smartdevicelink/sdl_evolution/issues/634

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with revisions. The revisions will include adding a flag in the policy table per app that describes if they are required to comply with the encryption or not as noted in [this comment.](https://github.com/smartdevicelink/sdl_evolution/issues/634#issuecomment-460710662)
There was additional discussion in regards to whether or not this encryption will be used in core from the policy table, essentially authentication and validation or if this flag needs to be passed from core into the proxies. It was determined that it is almost inferred by the proxies if they get an on-permission change with the new RPCs that are delegating they need to be encrypted and the app itself would need encryption, so core should send the on-permission changes to an app in regards to the RPC encryptions unless that app requires an encryption.   

Revisions were made on 2.19.19

SatbirTanda, 2019-07-01:
Every time the iOS library tries to start a secure service for RPC service type the HU (SYNC 3 18201_PRODUCT) responds with NACK,  can Luxoft look into this?

GetmanetsIrina, 2019-07-01:
@SatbirTanda, let's discuss your question internally. Please find a message from me in Slack.

GetmanetsIrina, 2019-07-02:
For now the issue `Every time the iOS library tries to start a secure service for RPC service type the HU (SYNC 3 18201_PRODUCT) responds with NACK` is under investigation, the root cause is not defined,  i'm waiting for SDL logs.
Found out that Satrib has the right environment and all required conditions for security service are taken into account.

GetmanetsIrina, 2019-07-04:
The issue `Every time the iOS library tries to start a secure service for RPC service type the HU (SYNC 3 18201_PRODUCT) responds with NACK` is resolved.

closed 2019-08-29 by JackLivio