smartdevicelink/sdl_core #2830: SDL sends unsucces general PerformIntereaction response to mobile app in case choiceID was selected

GetmanetsIrina, 2019-03-04:
### Bug Report
SDL sends unsucces general PerformIntereaction response to mobile app in case choiceID was selected

##### Reproduction Steps
1. App is registered and activated
2. ChoiceSet is added
3. Mobile app requests PerformIntereaction(BOTH, VR_ONLY)
4. Make choice( in case BOTH only VR choice )

##### Expected Behavior
SDL receives successful VR.PI with choiceID value and sends UI.ClosePopUp to UI.PI, HMI responds with unsuccessful UI.PI, then SDL sends successful response with choiceID to mobile app. 

##### Observed Behavior
SDL responds with unsuccessful result code and without choiceId to mobile app and does not send UI.ClosePopUp to HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.ford.com/SmartDeviceLinkMirror/sdl_core/commit/d85e7f2e5a844145149a480f8e5d2d8cdcc1a013
* Testing Against: HMI https://github.ford.com/SmartDeviceLinkMirror/sdl_hmi/commit/7962e1e858ca2c672fd53b1a1cd4ab08f59e4a32
Log: 
[SmartDeviceLinkCore.log.tar.gz](https://github.ford.com/SmartDeviceLinkMirror/sdl_core/files/6026/SmartDeviceLinkCore.log.tar.gz)

##### Diagrams
VR_ONLY mode:
[image]
BOTH mode:
[image]
ClosePopUp 
[image]

YarikMamykin, 2019-11-19:
@theresalech Please triage this issue and notice linked fix is ready for review.

theresalech, 2019-11-19:
@YarikMamykin both linked PRs (https://github.com/smartdevicelink/sdl_core/pull/3122, https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2293) have `This PR is **not ready** for review.` in their descriptions.  Please tag me and @jordynmackool on those PRs when they are confirmed as ready for review and descriptions have been updated.

YarikMamykin, 2019-11-22:
@theresalech please notice that PRs https://github.com/smartdevicelink/sdl_core/pull/3122 and https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2293 are **ready** for review.

iCollin, 2020-01-31:
Closed via #3122

PR #3122 (2020-01-31): Fix processing of VR only Perform Interaction response
Fixes #2830 #3112 #2990 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF test script attached to https://github.com/smartdevicelink/sdl_core/issues/2830

### Summary
In case of successful VR.PerformInteraction response from HMI with choiceID present in response
SDL doesn't send UI_ClosePopUp request to HMI if VR.PerformInteraction response came before UI.PerformInteraction response. More than that, SDL doesn't send successful response to mobile with choiceID.
Another problem is next: in c [... cut]
closed 2020-01-31 by iCollin