smartdevicelink/sdl_core #2840: [SDL 0173] Read Generic Network Signal Data

theresalech, 2019-03-12:
Proposal: [Read Generic Network Signal Data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0173-Read-Generic-Network-Signal-data.md)

> This proposal is to make vehicle data APIs (GetVehicleData/SubscribeVehicleData/UnsubscribeVehicleData/OnVehicleData) more flexible to allow additional new vehicle data items, while maintaining the security access via policies.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/502

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions. The author is to add a statement explaining how old core implementations using this method can be updated to expose new standardized vehicle data items. Additionally, the author is to add the three revisions noted in [this comment.](https://github.com/smartdevicelink/sdl_evolution/issues/502#issuecomment-465326262)

>The proposal .md was updated on March 11, 2019 to include the agreed upon revisions.

theresalech, 2019-07-24:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0173 Revisions - Read-Generic-Network-Signal-data](https://github.com/smartdevicelink/sdl_evolution/issues/768).  Please reference issue and proposal markdown file for more details.

theresalech, 2019-08-21:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0173 Revisions - Read Generic Network Signal Data Clarification](https://github.com/smartdevicelink/sdl_evolution/issues/806). Please reference issue and proposal markdown file for more details.

jacobkeeler, 2019-08-30:
Closing with the merge of #2977

theresalech, 2019-10-02:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [Revise 0173-Read-Generic-Network-Signal-data.md](https://github.com/smartdevicelink/sdl_evolution/issues/822). Please reference issue and proposal markdown file for more details.

JackLivio, 2019-10-04:
closed via #3062

PR #2977 (2019-08-30): [SDL 0173] - Read Generic Network Signal Data Implementation
Fixes #2840 

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
Manual testing is done with https://github.com/smartdevicelink/sdl_hmi/pull/210 
Automated testing is done with https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2217
Testing reports attached :  [RGNSD_test_reports.zip](https://github.com/smartdevicelink/sdl_core/files/3443042/RGNSD_test_reports.zip)

### Summary
Implementation of SDL evolution issue: https://github.com/smartdevicelink/sdl_evolution/issues/768 

 [... cut]
closed 2019-10-04 by JackLivio