smartdevicelink/sdl_core #2857: Core rejects RAI if unknown enums are present

jacobkeeler, 2019-03-22:
### Bug Report
Core by default rejects any message with `INVALID_DATA` if it contains any unknown enum values, but this should not be done for the RegisterAppInterface request, since this message is sent before version negotiation occurs

##### Reproduction Steps
1. Send a RegisterAppInterface request with a new AppHMIType (ex. "SHOPPING") and an existing AppHMIType (ex. "MEDIA")

##### Expected Behavior
Core sends a `SUCCESS` response with the negotiated version, which can be used by the application to determine that the new AppHMIType is not supported by the module

##### Observed Behavior
Core responds with `INVALID_DATA`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 5.0.0
* Testing Against: iOS test app

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

IGapchuk, 2020-05-20:
Hi @jacobkeeler ! I would like to propose to take a look at the proposal https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0298-Processing-of-unknown-enum-values-by-SDL-Core.md. I believe the proposal covers this issue and provides changes not only for RAI but for another PRCs too. Thank you!