smartdevicelink/sdl_core #2859: Core crash during ignition_off by ATF script execution

GetmanetsIrina, 2019-03-26:
### Bug Report
Core crash during ignition_off by ATF script execution

##### Reproduction Steps
1. Execute StartAndStopSDL.lua

##### Expected Behavior
SDL stops successfully

##### Observed Behavior
Sometimes core dump is observed during ignition_off

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
```

GetmanetsIrina, 2019-03-27:
@theresalech, please triage this issue

LuxoftAKutsan, 2019-03-27:
Backtrace with GDB : 
```cpp
(gdb) 
#0  0x00007fe596dc1428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007fe596dc302a in __GI_abort () at abort.c:89
#2  0x00007fe596e037ea in __libc_message (do_abort=2, fmt=fmt@entry=0x7fe596f1ced8 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007fe596e0a7c3 in malloc_printerr (ar_ptr=0x7fe597150b20 <main_arena>, ptr=0x4fa2b10, str=0x7fe596f19c92 "corrupted double-linked list", action=<optimized out>) at malloc.c:5006
#4  malloc_consolidate (av=av@entry=0x7fe597150b20 <main_arena>) at malloc.c:4175
#5  0x00007fe596e0c678 in _int_free (av=0x7fe597150b20 <main_arena>, p=<optimized out>, have_lock=0) at malloc.c:4075
#6  0x00007fe596e1053c in __GI___libc_free (mem=<optimized out>) at malloc.c:2968
#7  0x0000000002b9c1da in __gnu_cxx::new_allocator<std::shared_ptr<boost::asio::detail::posix_mutex> >::deallocate (this=0x478d3c0, __p=0x47a91b0) at /usr/include/c++/5/ext/new_allocator.h:110
#8  0x0000000002b9b2b8 in std::allocator_traits<std::allocator<std::shared_ptr<boost::asio::detail::posix_mutex> > >::deallocate (__a=..., __p=0x47a91b0, __n=41) at /usr/include/c++/5/bits/alloc_traits.h:517
#9  0x0000000002b99f40 in std::_Vector_base<std::shared_ptr<boost::asio::detail::posix_mutex>, std::allocator<std::shared_ptr<boost::asio::detail::posix_mutex> > >::_M_deallocate (this=0x478d3c0, 
    __p=0x47a91b0, __n=41) at /usr/include/c++/5/bits/stl_vector.h:178
#10 0x0000000002b98edd in std::_Vector_base<std::shared_ptr<boost::asio::detail::posix_mutex>, std::allocator<std::shared_ptr<boost::asio::detail::posix_mutex> > >::~_Vector_base (this=0x478d3c0, 
    __in_chrg=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:160
#11 0x0000000002b9813f in std::vector<std::shared_ptr<boost::asio::detail::posix_mutex>, std::allocator<std::shared_ptr<boost::asio::detail::posix_mutex> > >::~vector (this=0x478d3c0, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/stl_vector.h:425
#12 0x0000000002b9790f in boost::asio::ssl::detail::openssl_init_base::do_init::~do_init (this=0x478d3c0, __in_chrg=<optimized out>)
    at /home/developer/sdl/3rd_party/include/boost/asio/ssl/detail/impl/openssl_init.ipp:61
#13 0x0000000002b9e784 in std::_Sp_counted_ptr<boost::asio::ssl::detail::openssl_init_base::do_init*, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x47ac740) at /usr/include/c++/5/bits/shared_ptr_base.h:374
#14 0x00000000020207d4 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x47ac740) at /usr/include/c++/5/bits/shared_ptr_base.h:150
#15 0x000000000202044b in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x33166c8 <boost::asio::ssl::detail::openssl_init_base::instance()::init+8>, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/shared_ptr_base.h:659
#16 0x0000000002b9792e in std::__shared_ptr<boost::asio::ssl::detail::openssl_init_base::do_init, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (
    this=0x33166c0 <boost::asio::ssl::detail::openssl_init_base::instance()::init>, __in_chrg=<optimized out>) at /usr/include/c++/5/bits/shared_ptr_base.h:925
#17 0x0000000002b9794a in std::shared_ptr<boost::asio::ssl::detail::openssl_init_base::do_init>::~shared_ptr (this=0x33166c0 <boost::asio::ssl::detail::openssl_init_base::instance()::init>, 
    __in_chrg=<optimized out>) at /usr/include/c++/5/bits/shared_ptr.h:93
#18 0x00007fe596dc5ff8 in __run_exit_handlers (status=0, listp=0x7fe5971505f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#19 0x00007fe596dc6045 in __GI_exit (status=<optimized out>) at exit.c:104
#20 0x00007fe596dac837 in __libc_start_main (main=0x201c2d6 <main(int32_t, char**)>, argc=1, argv=0x7ffdc235c7c8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffdc235c7b8)
    at ../csu/libc-start.c:325
#21 0x000000000201c209 in _start ()

```

theresalech, 2019-03-27:
We are noticing this crash on master as well.  As we do not believe this to be a new issue introduced by Core 5.1, we will need to note it as something to investigate after the Core 5.1 release, if agreed to by the SDLC Steering Committee.

LuxoftAKutsan, 2019-04-05:
@theresalech  Luxoft recommend to fix this issue in scope of next 5.1.1 stabilization release due to reasons:
  - to have more stable SDL
  - to not have 'stack trace' error messages during automation testing

JackLivio, 2019-10-14:
This issue was resolved in the 6.0.0 development and is no longer reproducible. 

Potentially fixed by #2939, #3048, #3022, or #3029

closed 2019-10-14 by JackLivio