smartdevicelink/sdl_core #2860: SDL build fails in case if ENABLE_LOG=OFF or ENABLE_SECURITY=OFF

dboltovskyi, 2019-03-27:
### Bug Report
SDL build fails in case if either `ENABLE_LOG=OFF` or `ENABLE_SECURITY=OFF` cmake parameter is defined

##### Reproduction Steps
1. Try to build SDL with `ENABLE_LOG=OFF` or/and `ENABLE_SECURITY=OFF` cmake parameter

##### Expected Behavior
SDL build finished successfully

##### Observed Behavior
SDL build fails

##### OS & Version Information
* OS/Version: Ubuntu 16.04 amd64
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0 (c647e94)

##### Test Case, Sample Code, and / or Example App
[log_ENABLE_LOG_OFF.txt](https://github.com/smartdevicelink/sdl_core/files/3013141/log_ENABLE_LOG_OFF.txt)
[log_ENABLE_SECURITY_OFF.txt](https://github.com/smartdevicelink/sdl_core/files/3013165/log_ENABLE_SECURITY_OFF.txt)

GetmanetsIrina, 2019-03-27:
@theresalech. please triage this issue

jacobkeeler, 2019-03-27:
Closing with the merge of #2861

dboltovskyi, 2019-03-28:
Please notice issue `ENABLE_LOG=OFF` is still reproducible on a https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0 (91d2c60)

jacobkeeler, 2019-03-28:
It was not specified in the issue, but this last failure only occurs when ENABLE_LOG=OFF and BUILD_TESTS=ON

jacobkeeler, 2019-03-29:
Fully resolved with the merge of #2863

dboltovskyi, 2019-04-01:
Please notice issue is reopened since it's still reproduced on a clean environment on a last SDL commit (abdfa1d)
Exact cmake command: `cmake <path_to_sdl_core> -DENABLE_LOG=OFF`
Error is the same

jacobkeeler, 2019-04-01:
@dboltovskyi Hmmm, the same failure does not happen on my machine, I used Ubuntu 16.04 with a clean build and 3rd party libs removed so that they would need to be installed. I ran into a few issues this way, but none of them were the exact error displayed in the log. I've submitted #2870 for the issues that did appear on my machine, I can perform a completely clean build without issue after these changes. Please verify whether or not this fixes the issues on your machine.

dboltovskyi, 2019-04-02:
@jacobkeeler 
It seems I found the difference between build environments.
We used two define environment variables `THIRD_PARTY_INSTALL_PREFIX` and `THIRD_PARTY_INSTALL_PREFIX_ARCH` to some folder.
They allow us to build all 3rd party libraries to this defined folder.
If these variables are not set, all 3rd party libraries will be built to system folders `/usr/local/lib` and `/usr/local/include`

So the issue is reproduced only in case if these environment variables are defined.
I believe @LuxoftAKutsan could give more information on this.

LuxoftAKutsan, 2019-04-02:
By default SDL is trying to install third party libraries into /usr/loca/lib. This requires sudo access.
To avoid sudo access for build  `THIRD_PARTY_INSTALL_PREFIX` and `THIRD_PARTY_INSTALL_PREFIX_ARCH` can be used. Now in case if this variables are not empty and  ` ENABLE_LOG=OFF` compilation issues reproduces.

jacobkeeler, 2019-04-03:
Alright, I believe that I've finally fixed the issue fully with #2876

dboltovskyi, 2019-04-04:
@jacobkeeler Fix works. Thank you a lot.

jacobkeeler, 2019-04-04:
Closing with the merge of #2876

PR #2861 (2019-03-27): Fix build failures depending on options

Fixes #2860 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run build with `ENABLE_SECURITY=OFF` and `ENABLE_LOG=OFF`

### Summary
Fix build issues when `ENABLE_SECURITY=OFF` or `ENABLE_LOG=OFF`

### Changelog
##### Bug Fixes
* Fixes build issues when `ENABLE_SECURITY=OFF` or `ENABLE_LOG=OFF`

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #2863 (2019-03-29): Fix build when ENABLE_LOG=OFF and BUILD_TESTS=ON
Fixes #2860 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Build with ENABLE_LOG=OFF and BUILD_TESTS=ON

### Summary
Forcing `-Werror` has consistently resulted in a fragile build when building with different options, this PR changes some of these error back into warnings so that false positives do not result in build failures. Also fixes linking issue with the Low Voltage signal handler library tests.

### Changelog
##### Bug Fixes
* Remove unused variable warnings from `-Werror`, to prevent iss [... cut]
PR #2870 (2019-04-01): Fix missing 3rd-party dependencies in CMake files
Fixes #2860 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run build with `ENABLE_LOG=OFF`

### Summary
Fixes issues with 3rd party dependencies in several components, causing a build failure on the first attempt to install

### Changelog
##### Bug Fixes
* Fixes issues with 3rd party dependencies in several components, one of which was removed without reason in https://github.com/smartdevicelink/sdl_core/commit/376c972ba76aa042fbd17f36cce9518d57d7b52d

### CLA
- [x] I have signed [the CLA](https [... cut]
PR #2876 (2019-04-04): Add missing Boost include directories

Fixes #2860 (hopefully)

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Set `THIRD_PARTY_INSTALL_PREFIX` and `THIRD_PARTY_INSTALL_PREFIX_ARCH` in your environment and build with `ENABLE_LOG=OFF`

### Summary
The BOOST_INCLUDE_DIR was not properly scoped in the 3rd-party CMakeLists, and this value was not included in the include directories of many components that needed it. This PR fixes those issues. No issues occurred with ENABLE_LOG=ON because often the LOG4CXX_INCLUDE_DIRECTORY was the same as the BOOST_ [... cut]
closed 2019-04-04 by jacobkeeler