smartdevicelink/sdl_core #2864: Unable to update already published App Service

dboltovskyi, 2019-04-01:
### Bug Report
Unable to update already published App Service

##### Reproduction Steps
1. App publish a particular service 
2. The same app publish the same service with updated manifest

##### Expected Behavior
SDL sends `OnSystemCapabilityUpdated(MANIFEST_UPDATE)`

##### Observed Behavior
SDL sends `OnSystemCapabilityUpdated(PUBLISHED)`

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [5.1.0](https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0) (abdfa1d)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-01_13-34-09.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3028904/2019-04-01_13-34-09.tar.gz)
[2019-04-01_13-36-13.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3028906/2019-04-01_13-36-13.tar.gz)

dboltovskyi, 2019-04-01:
@theresalech, please triage this issue

JackLivio, 2019-04-01:
@dboltovskyi Thank you for bringing this up. Unfortunately  the app services proposal does not describe ways to update or remove services. We are looking at defining this behavior in a new proposal to take through the evolution process in the future.

LuxoftAKutsan, 2019-04-05:
@JackLivio Any way original proposal contains `MANIFEST_UPDATE` `ServiceUpdateReason`. I believe that is should be makred as not implemented in HMI_API ad in proposal untill it won't be implemented.

JackLivio, 2019-04-05:
Ok thank you for the note. We will mark as not implemented for now and target the implementation for post 5.1

closed 2019-04-09 by JackLivio