smartdevicelink/sdl_core #2865: SDL sends GENERIC_ERROR response instead of TIMED_OUT in case App Service doesn't respond within defined timeout

dboltovskyi, 2019-04-01:
### Bug Report
SDL sends GENERIC_ERROR response instead of TIMED_OUT in case App Service doesn't respond within defined timeout

##### Reproduction Steps
1. App_1 publishes a particular service
2. App_2 is requested `GetAppServiceData` RPC for this service type
3. App_1 does not provide response within defined timeout

##### Expected Behavior
SDL sends `TIMED_OUT` response to App_2

##### Observed Behavior
SDL sends `GENERIC_ERROR` response to App_2

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [5.1.0](https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0) (abdfa1d)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-01_13-42-59.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3028942/2019-04-01_13-42-59.tar.gz)

dboltovskyi, 2019-04-01:
@theresalech, please triage this issue

JackLivio, 2019-04-01:
@dboltovskyi For this case we followed the existing behavior seen when the HMI does not respond to a request from the mobile device in the set timeout. In this event, the mobile device will receive GENERIC_ERROR, not TIMED_OUT. 

If we want to change the behavior for app services to return TIMED_OUT, then we would also need to do so for the rest of Mobile->HMI requests and responses.

I understand that GENERIC_ERROR over TIMED_OUT is not the obvious choice at first glance.

LuxoftAKutsan, 2019-04-05:
@JackLivio Is it going to be delivered as a revision into original proposal?

jacobkeeler, 2019-04-05:
@LuxoftAKutsan The only `TIMED_OUT` response I see referenced in the proposal is with regards to RPC passthrough via handledRPCs (which would not apply to the case with `GetAppServiceData` as described in this issue).

With regards to RPC passthrough, an issue came up where we realized that some of the potential handled RPCs might not always be handled fully by an app service (such as `ButtonPress`). As such, a waterfall setup was created for handling these messages:
>
> Flow 1 (normal result code)
> 1. app1 sends an RPC which is handled by an active service
> 2. Core sees that an active service includes this function ID in it's handledRPCs, passes the raw message to that service provider
> 3. The service provider responds with any result code other than UNSUPPORTED_REQUEST
> 4. Core passes the RPC response to app1
>
> Flow 2 (result code UNSUPPORTED_REQUEST)
> 1. app1 sends an RPC which is handled by an active service
> 2. Core sees that an active service includes this function ID in it's handledRPCs, passes the raw message to that service provider
> 3. The service provider responds with result code UNSUPPORTED_REQUEST (indicating that it cannot handle some part of the request) or times out
> 4. Repeat steps 2 and 3 until a response is received with a result code other than UNSUPPORTED_REQUEST (in which case, refer to Flow 1, Step 4)
> 5. If all active services that handle this function ID have been queried and responded with UNSUPPORTED_REQUEST, handle the RPC normally (passing it through to Core)

As such, the `TIMED_OUT` response doesn't really apply anymore to this flow, as a timed out request will just result in the original request being passed on to the next service in the list.

I also don't think that which result code is returned for this case is a particularly important detail. `GENERIC_ERROR` is what is currently used for RPC timeouts, and so if we want to change this, we should do so for all RPC requests, rather than make an exception for this feature.

closed 2019-04-12 by jacobkeeler