smartdevicelink/sdl_core #2866: SDL dos not send response to HMI in case App Service doesn't respond within defined timeout

dboltovskyi, 2019-04-01:
### Bug Report
SDL dos not send response to HMI in case App Service doesn't respond within defined timeout

##### Reproduction Steps
1. App_1 published a particular service
2. HMI is requested `AppService.GetAppServiceData` RPC for this service type
3. App_1 does not provide response within defined timeout

##### Expected Behavior
SDL sends `TIMED_OUT` response to HMI

##### Observed Behavior
SDL does not send response to HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [5.1.0](https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0) (abdfa1d)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-01_13-48-30.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3028965/2019-04-01_13-48-30.tar.gz)

dboltovskyi, 2019-04-01:
@theresalech, please triage this issue

jacobkeeler, 2019-04-02:
Fixed with the merge of #2871

PR #2871 (2019-04-02): Add timeout logic for HMI App Service Consumer RPCs
Fixes #2866 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Run [013_H2M_GENERIC_ERROR.lua](https://github.com/smartdevicelink/sdl_core/files/3031216/013_H2M_GENERIC_ERROR.txt)

### Summary
Add missing timeout response for `GetAppServiceData` and `PerformAppServiceInteraction` HMI requests

### Changelog
##### Bug Fixes
* Add missing timeout response for `GetAppServiceData` and `PerformAppServiceInteraction` HMI requests
* Fix issue where `code` was incorrectly set in `RequestFromHMI::SendRe [... cut]
closed 2019-04-02 by jacobkeeler