smartdevicelink/sdl_core #2867: SDL allows HMI to be App Service consumer however unsubscription is not working

dboltovskyi, 2019-04-01:
### Bug Report
SDL allows HMI to be App Service consumer however unsubscription is not working

##### Reproduction Steps
1. App_1 published a particular service
2. HMI unsubscribes to get AppServiceData
3. App_1 sends `OnAppServiceData` notification 

##### Expected Behavior
SDL does not send Notification to HMI

##### Observed Behavior
SDL does send Notification to HMI

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [5.1.0](https://github.com/smartdevicelink/sdl_core/tree/release/5.1.0) (abdfa1d)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-01_14-09-16.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3029037/2019-04-01_14-09-16.tar.gz)

dboltovskyi, 2019-04-01:
@theresalech, please triage this issue

jacobkeeler, 2019-04-01:
@dboltovskyi We are not considering this to be a bug, due to the fact that notification subscription is handled through the message broker on the HMI side. The `subscribe` parameter will not do anything if sent by the HMI, but it can still be received by the HMI if it is a service provider. Since the message is bidirectional, this parameter should still remain present in the HMI_API variant of the `GetAppServiceData` message.

closed 2019-04-01 by jacobkeeler

LuxoftAKutsan, 2019-04-05:
@jacobkeeler It is not obvious from original proposal that : 
>> The subscribe parameter will not do anything if sent by the HMI

Propose to include it to proposal and to HMI_API parameter description