smartdevicelink/sdl_core #2872: SDL does not take into account EmbeddedServices parameter of .INI file

dboltovskyi, 2019-04-02:
### Bug Report
SDL does not take into account `EmbeddedServices` parameter of `.INI` file

##### Reproduction Steps
1. Change the values for `EmbeddedServices` parameter of `.INI` file
2.  Try to publish app service from mobile or from HMI with the type defined by parameter

##### Expected Behavior
According to proposal services defined by this parameter have to be used as default and will only become primary service publisher with direct user interaction.

##### Observed Behavior
App services becames primary even if they are defined in `EmbeddedServices` parameter

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 5.1.0 https://github.com/smartdevicelink/sdl_core/commit/1265fce0ec352f4a2a65709960696a77bd4e68fb
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
<TBA>

dboltovskyi, 2019-04-02:
@theresalech please triage this issue

jacobkeeler, 2019-04-02:
@dboltovskyi The `EmbeddedServices` parameter is only taken into account if the HMI actually publishes a service of that type, in which case it becomes the default service and is immediately activated. It doesn't really make any sense to restrict app services in this way if the HMI doesn't publish the services that are included in this parameter.

dboltovskyi, 2019-04-04:
@jacobkeeler 
Following described logic let's do the following **steps**:
1. Update `EmbeddedServices` parameter in `.INI` by excluding `MEDIA` service
2. Run [PublishAppService/002_HMI_success_flow.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/AppServices/PublishAppService/002_HMI_success_flow.lua) script and check results

**Expected:**
Service is not activated automatically 

**Actual:**
Service becomes activated

So the parameter is not taken into account. Suggest please is this correct behavior?

jacobkeeler, 2019-04-04:
There are a few rules that take effect here:
1. The first service of it's type is activated when it is published.
2. If a given service type is in the `EmbeddedServices` parameter, the embedded service is made the default service for that type when it is published.
3. The default service for a given type is activated immediately when it is published.

So in the case you pointed out, this service is still activated because of rule 1.

dboltovskyi, 2019-04-04:
@jacobkeeler 
Thank you for explanation
I'll try to check other rules and their combinations

dboltovskyi, 2019-04-08:
Checks passed

closed 2019-04-08 by dboltovskyi