smartdevicelink/sdl_core #2873: SDL sends INVALID_DATA and GENERIC_ERROR  responses instead of SUCCESS in case Mobile App sends SystemRequest (ICON_URL) with binary data

VjKlepikov, 2019-04-03:
[Delete any non-applicable sections, but we may ask for more information.]

### Bug Report
[Summary]

Precondition:
  1. Cloud App in the policy table
  2. HMI and SDL are started
  3.  Mobile App is registered and activated

##### Reproduction Steps
1. [Send SystemRequest (ICON_URL) with binary data to SDL]

##### Expected Behavior
[SDL sends successful response to Mobile App]

##### Observed Behavior
[SDL sends INVALID_DATA and GENERIC_ERROR  responses to Mobile App

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

jacobkeeler, 2019-04-03:
This request type should not be used without a corresponding `OnSystemRequest` of `ICON_URL` type sent by Core to that application. 

In your logs, I see that an `OnSystemRequest(ICON_URL)` with a url of `https://noiconurl.org` is sent by Core, but the fileName sent in the `SystemRequest(ICON_URL)` is `icon.png`. 

This parameter needs to be filled with the original url sent in the `OnSystemRequest(ICON_URL)`, because multiple `OnSystemRequest(ICON_URL)` notifications can be sent in a session, and without the original URL it is not possible for Core to differentiate these requests.

jacobkeeler, 2019-04-03:
It appears that the extra GENERIC_ERROR response is just because of a missing return after the INVALID_DATA response, will submit a fix for this.

dboltovskyi, 2019-04-03:
@theresalech Please triage this issue

jacobkeeler, 2019-04-04:
Closing with the merge of #2875

PR #2875 (2019-04-04): Fix Icon Url Corner Cases
Fixes #2873 and #2874 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Send invalid icon file name. Only observe INVALID DATA response and no GENERIC_ERROR.
Send correlation ID -1 and observe INVALID_ID response.

### Summary
- Add missing return value after sending invalid data. 
- Checks if correlation id is valid. Otherwise sends INVALID_ID.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-04-04 by jacobkeeler