smartdevicelink/sdl_core #2874: SDL crash happens when receiving SystemRequest (ICON_URL)  with correlationID -1 (INVALID_ENUM)

VjKlepikov, 2019-04-03:
### Bug Report
SDL crash happens when receiving SystemRequest (ICON_URL)  with correlationID: -1 (INVALID_ENUM)

Precondition:

   1. Cloud App in the policy table
   2. HMI and SDL are started
   3. Mobile App is registered and activated

Reproduction Steps

    Send SystemRequest (ICON_URL) with correlationID: -1 

Expected Behavior

     No any failures

Observed Behavior

     Core crash

OS & Version Information

    OS/Version: [16.04]
    SDL Core Version: [5.1.0] b9500e5

Test Case, Sample Code, and / or Example App

[Paste a link to a PR, gist, or other code that exemplifies this behavior]

dboltovskyi, 2019-04-03:
@theresalech Please triage this issue

ShobhitAd, 2019-04-04:
Closing with merge of https://github.com/smartdevicelink/sdl_core/pull/2875

closed 2019-04-04 by ShobhitAd