smartdevicelink/sdl_core #2877: SDL doesn't take into account functional group permissions for Consumer App

dboltovskyi, 2019-04-04:
### Bug Report
SDL doesn't take into account functional group permissions for Consumer App that is defined in Policy Table

##### Reproduction Steps
1. App_1 (Service Provider) and App_2 (Consumer) are registered on SDL.
2. AppServiceProvider permissions(with NAVIGATION AppService permissions to handle rpc `SendLocation`) are assigned for App_1
3. SendLocation permissions **are NOT assigned** for App_2
4. App_1 sends a PublishAppService (with {serviceType=NAVIGATION, handledRPC=SendLocation} in the manifest)
5. App_2 sends a SendLocation request to Core

##### Expected Behavior
1. Core does not forward the request to App_1
2. Core does not forward the request to HMI
3. Core responds to App_2 with { success = false, resultCode = "DISALLOWED" }

##### Observed Behavior
1. Core forwards the request to App_1
2. Core does not forward the request to HMI
3. Core responds to App_2 with { success = true, resultCode = "SUCCESS" } (In case if App_1 responds to request successfully)

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: [5.1.0](https://github.com/smartdevicelink/sdl_core/commit/b9500e5e6f1c39db7eaee3a7416179edf0ac5dc6)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-04_16-42-41.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3043764/2019-04-04_16-42-41.tar.gz)

dboltovskyi, 2019-04-04:
It seems SDL doesn't take into account other restrictions that could be defined in Policy Table
Such as:
  - HMI type of the application: if some app tries to request any of Remote Control RPCs and `AppHMIType` of this app is not `REMOTE_CONTROL` this request will be rejected by SDL (see [./test_scripts/RC/CLIMATE_RADIO/GetInteriorVehicleData/004_Disallow_flow_in_case_appHMIType_is_not_REMOTE_CONTROL.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/RC/CLIMATE_RADIO/GetInteriorVehicleData/004_Disallow_flow_in_case_appHMIType_is_not_REMOTE_CONTROL.lua)** )
  - RC Module Type: if some app tries to request RPC with module type that is not allowed for this app this request will be rejected by SDL (see [./test_scripts/RC/CLIMATE_RADIO/ButtonPress/002_disallow_flow_by_policy_for_climate.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/RC/CLIMATE_RADIO/ButtonPress/002_disallow_flow_by_policy_for_climate.lua) )

jacobkeeler, 2019-04-04:
@dboltovskyi The approach we took with this feature was that the OEM should not necessarily dictate which messages should be sent during inter-app communication, and thus normal policy restrictions do not apply to these messages. Just because an OEM doesn't allow a `SendLocation` to it's own module doesn't mean that an service handling `SendLocation` shouldn't be able to receive this message. In addition, the normal policy check could not be performed because this would disallow all future RPCs, which this feature is designed to support.

LuxoftAKutsan, 2019-04-05:
@jacobkeeler  it is not clear from the proposal that policies will be ignored if connected to SDL AppService handles certain RPC's. I would say that this solution should be discussed. 
Potentially it can be some kind of security issue.

LuxoftAKutsan, 2019-04-05:
From Luxoft and Ford (discussed with @robinmk @stefanek22 ) perspective Policies should not be avoided for RPCs that SDL pass throw to AppServices. 
Any mention about ignoring policies are available in original proposal, so it was not discussed during review. 
It changes SDL Policies model, adds additional no obvious use cases, and potentially can be security issue.

JackLivio, 2019-04-05:
I am open to discussion for a solution here. But first I would like to explain the reasoning behind this implementation detail. 

The app services features allows for the pass through of RPCs from app-> app that are unknown to Core's mobile api and policy implementation. If we wanted to enforce  policies for the RPC pass through feature, then we would need to create a way for Core to enforce permissions on RPCs that it does not have a RPC name -> function ID mapping for. This type of change would require major changes to both Core and to the SDL server, which were not outlined in the proposal or possible for a 5.1 release. 

It would be possible for the pass through feature to be able to check permissions for function ids that Core recognizes, but that solution would arguably be as insecure as checking no permissions. Also I think this approach would only be a temporary bandaid and not an actual solution

Here are three possible options I can think of so far aside from leaving the implementation as is. 

1). Overhaul SDL Core and Policy Server to pass a RPC name -> function ID mapping in the policy table update. This mapping can be used by Core to establish permissions for function IDs it does not have in its APIs.
  - Pros: Most secure and fully encompassing solution.
  - Cons: Likely a breaking change. Require refactor to policies. Will take time and proposal entries.

2). Only check permissions for known RPCs.
  - Pros: Quick fix
  - Cons: Not a real solution once new RPCs are added to the API.

3). Add new parameter to app policies section, `allowPassThrough`. If an app sends an rpc that will reach a provider, Core could check if this field is true to allow the pass through to happen. If it is false, then Core will respond disallowed. This would be more of an all or nothing permission type of approach. At least this way an OEM could revoke the permissions in case of a bad actor.
  - Pros: Quick Fix
  - Cons: Low resolution of control.

LuxoftAKutsan, 2019-04-05:
@JackLivio OK, I can see implicit mention of this functionality in proposal : 
```xml
<param name="handledRPCs" type="Integer" array="true" mandatory="false">
			<description> 
...
This parameter is an integer to allow for new function IDs to be used by apps on older versions of SDL Core.</description>
</param>
```

My opinion that it should be some mix between 3rd and second proposed solution.

Policies should contain some flag that can enforce strict policy check according existing API, and all unknown functionID's will be DISALLOWED. 
But even if flag will allow unknown FunctionID's to for oldVersions of sdl, some policy rules should be checked any way : revoked app, not consented device, etc ...

JackLivio, 2019-04-05:
@LuxoftAKutsan I like your suggestion of using a combination for now. We can get started on developing this as the solution but including this in the 5.1 release might come at the cost of having to extend the 5.1 release date again.

LuxoftAKutsan, 2019-04-08:
@JackLivio From Luxoft and Ford(@robinmk, Stefan Bankowski) perspective fix for this issue should be included in the 5.1 release considering the cost of having to extend the 5.1 release date again.

LuxoftAKutsan, 2019-04-11:
@theresalech  when all related PRs are planned to be merged to the release branch?

JackLivio, 2019-04-11:
Fixed by #2881

closed 2019-04-11 by JackLivio

theresalech, 2019-04-11:
@LuxoftAKutsan All related ATF PRs have been merged into the release branch.

PR #2881 (2019-04-09): App Services Allow Unknown RPC Passthrough
Fixes #2877 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Add new field to application policies in sdl_preloaded_pt.json
 > "allow_unknown_rpc_passthrough" : true

### Summary
Adds new field to policies. Field will allow/disallow an app consumer from performing an unknown rpc request that is passed through to another app service.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)