smartdevicelink/sdl_core #2878: Resumption.Application data is not restored on HMI after unexpected disconnect and connect are performed

HSavynetska, 2019-04-05:
### Bug Report
Application data is not restored on HMI after unexpected disconnect and connect are performed

##### Reproduction Steps
1.SDL and HMI are started
2. App1is registered and activated 
3. App1 added AddCommand
4. Unexpected disconnect and connect are performed

##### Expected Behavior
SDL sends data for resuming after sending "BasicCommunication.OnAppRegistered" to the HMI

##### Observed Behavior
SDL sends data for resuming before sending "BasicCommunication.OnAppRegistered" to the HMI and On HMI does not display all data for applications

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: current develop 12c02a9
* Testing Against: SPT App Version 20190122-Nightly-Android

## Attached 
[sdl.log and hmi.log.zip](https://github.ford.com/SmartDeviceLinkMirror/sdl_core/files/7330/sdl.log.and.hmi.log.zip)

LuxoftAKutsan, 2019-04-05:
This is race condition issue between SDL and HMI communication.
BasicCommunication and VR(or UI any other interface) have 2 different connections.

SDL sends BasicCommunication,OnAppRegistered, and right after it send UI.AddCommand.
There is no guarantees that BasicCommunication,OnAppRegistered notification will be received and processed by HMI before UI.AddCommand.

Issues can be fixed on HMI side - connect all interfaces by one connection, that will guarantee sequential processing.
But some OEM's may have similar issues.
So the problem is that SDL allow different channels for different interfaces.

I think that this problem should be resolved with proposal.
@JackLivio what do you think?

JackLivio, 2019-04-05:
I do not think that this change is needed to be made in Core to achieve sequential processing. In the Generic HMI we use one web socket to receive messages for all of the interfaces from core. The generic HMI then routes the messages to the appropriate HMI components. I believe this ensures that these messages are received by the HMI in a guaranteed order?

Please reference the [Controller.js](https://github.com/smartdevicelink/generic_hmi/blob/master/src/js/Controllers/Controller.js#L24)

LuxoftAKutsan, 2019-04-12:
@JackLivio  should we mention in hmi_guidelines this issue? 
I suppose that guidelines should contain recommendation : use all HMI interfaces by one connection. 
Such bugs on OEM side may be rely not obvious. But in case if we will add to guidelines why should be keep ability to use multiple connection for each HMI interface in sdl_core?

iCollin, 2020-11-02:
@HSavynetska @LuxoftAKutsan I cannot reproduce this issue after merge of 0190. 

Can you please confirm if this issue is still actual?

dboltovskyi, 2020-11-02:
@iCollin Please notice we can't reproduce this issue as well. 
It's very likely it was fixed within https://github.com/smartdevicelink/sdl_hmi/pull/191 

Issue is closed.

closed 2020-11-02 by dboltovskyi