smartdevicelink/sdl_core #2879: SDL proceed with ButtonPress for Consumer App if it does not correspond to RC module

dboltovskyi, 2019-04-05:
### Bug Report
SDL proceed with ButtonPress for Consumer App if button name does not correspond to RC module

##### Reproduction Steps
1. App_1 is Service Provider:
  * AppServiceProvider permissions(with `MEDIA` AppService permissions to handle rpc ButtonPress)
  * App_1 published App Service (with {serviceType=MEDIA, handledRPC=ButtonPress} in the manifest)
2. App_2 is a Consumer, which is REMOTE_CONTROL app, registered on SDL
3. App_2 sends a ButtonPress request with button which does not correspond to `CLIMATE` module (e.g. `VOLUME_UP`)

##### Expected Behavior
SDL transfer request to App Service provider (App_1)

##### Observed Behavior
SDL respond with { success=false, resultCode=INVALID_DATA }

##### Notes:
This is App Services modification of existing RC test case described in [./test_scripts/RC/CLIMATE_RADIO/ButtonPress/005_not_matached_buttonName_and_moduleType_for_radio_and_climate.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/RC/CLIMATE_RADIO/ButtonPress/005_not_matached_buttonName_and_moduleType_for_radio_and_climate.lua)

##### OS & Version Information
* OS/Version: [The OS/Version you were running when the bug occurred]
* SDL Core Version: [The version of SDL Core you were using when the bug occurred]
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
[2019-04-05_16-33-43.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3048147/2019-04-05_16-33-43.tar.gz)

dboltovskyi, 2019-04-05:
@theresalech Please triage this issue

JackLivio, 2019-04-05:
Regarding button press and pass through please note @jacobkeeler [comment in this issue](https://github.com/smartdevicelink/sdl_core/issues/2865#issuecomment-480315129).

> With regards to RPC passthrough, an issue came up where we realized that some of the potential handled RPCs might not always be handled fully by an app service (such as `ButtonPress`). As such, a waterfall setup was created for handling these messages:
>
> Flow 1 (normal result code)
> 1. app1 sends an RPC which is handled by an active service
> 2. Core sees that an active service includes this function ID in it's handledRPCs, passes the raw message to that service provider
> 3. The service provider responds with any result code other than UNSUPPORTED_REQUEST
> 4. Core passes the RPC response to app1
>
> Flow 2 (result code UNSUPPORTED_REQUEST)
> 1. app1 sends an RPC which is handled by an active service
> 2. Core sees that an active service includes this function ID in it's handledRPCs, passes the raw message to that service provider
> 3. The service provider responds with result code UNSUPPORTED_REQUEST (indicating that it cannot handle some part of the request) or times out
> 4. Repeat steps 2 and 3 until a response is received with a result code other than UNSUPPORTED_REQUEST (in which case, refer to Flow 1, Step 4)
> 5. If all active services that handle this function ID have been queried and responded with UNSUPPORTED_REQUEST, handle the RPC normally (passing it through to Core)

Also I am a little confused about the issue title and your expected vs observed behavior. Did you have expected behavior and observed behavior reversed?

JackLivio, 2019-04-05:
Also because of the nature of the pass through feature: if VOLUME_UP and module type CLIMATE are passed to a provider, it is up to the provider to recognize the mismatch and respond with INVALID_DATA. 

This is because the pas through feature needs to support RPCs and RPC params that are unknown to core. It will be up to the provider to do some message validation on its own that Core will not be able to handle.

dboltovskyi, 2019-04-08:
@JackLivio 
Thank you for noticing mismatch between expected and observed behavior. I've updated the description.
And also thank you for clarification.

closed 2019-04-08 by dboltovskyi