smartdevicelink/sdl_core #2883: SDL doesn't store cloud_transport_type parameter in Policy DB for EXTERNAL_PROPRIETARY mode

dboltovskyi, 2019-04-09:
### Bug Report
SDL doesn't store `cloud_transport_type` parameter in Policy DB for `EXTERNAL_PROPRIETARY` mode

##### Reproduction Steps
0. Make sure SDL is built in `EXTERNAL_PROPRIETARY` mode
1. Start SDL, HMI, connect Mobile, register new app 
  => PTU is started
2. Make sure PTU brings new cloud app with related parameters incl. `cloud_transport_type`
3. Check parameters for cloud app by sending `GetCloudAppProperties` request and check response
  => `cloudTransportType` is not empty and equals to `cloud_transport_type`
4. Perform Ignition Off / On
5. Start SDL, HMI, connect Mobile, register new app 
6. Send `GetCloudAppProperties` request again

##### Expected Behavior
In response there is non-empty `cloudTransportType` parameter

##### Observed Behavior
In response there is no `cloudTransportType` parameter

##### Notes
Issue is not reproduced if SDL is built in `PROPRIETARY` mode 

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version:  [5.1.0](https://github.com/smartdevicelink/sdl_core/commit/53fc68a)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
[2019-04-09_15-56-12.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3059087/2019-04-09_15-56-12.tar.gz)

dboltovskyi, 2019-04-09:
@theresalech Please triage this issue

JackLivio, 2019-04-09:
Fixed by #2881

closed 2019-04-09 by JackLivio