smartdevicelink/sdl_core #2884: Wrong response of rpc "ReadDID" from SDL to App

Nilesh-harman, 2019-04-11:
I find an issue related to rpc ReadDID. When I send a rpc "ReadDID" request from SPT app with invalid ECU number to SDL then HMI reply back to SDL with resultCode UNSUPPORTED_RESOURCE" with info "The Requested ECU is not Available" with error code :2. SDL reply back to app with the same resultCode and info but with "success" : true.
As per SDL integration guideline, it should be "Failure". 

From the SDL 5.1 project, In file "read_did_request.cc"  in function "ReadDIDRequest::on_event" the called "PrepareResultForMobileResponse" returns true so response is "success".
i.e. 
In function bool CommandRequestImpl::PrepareResultForMobileResponse(hmi_apis::Common_Result::eType result_code,
    HmiInterfaces::InterfaceID interface) const {

  if ((hmi_apis::Common_Result::UNSUPPORTED_RESOURCE == result_code) &&
      (HmiInterfaces::STATE_NOT_AVAILABLE != state)) {
    return true;
  }

Can you please help me, why we are returning true here?
  
  
Please refer below for the App-SDL-HMI communcation:

From App to SDL:
=====================
[application_manager_impl.cc:2124] :ConvertSOtoMessage::Convertion result: { "id" : 54, "jsonrpc" : "2.0", "method" : "VehicleInfo.ReadDID", "params" : { "appID" : 1329276570, "didLocation" : [ 56832 ], "ecuName" : 2000 } }

From SDL to HMI:
=====================
Handle::FROM_SDL [{ "id" : 54, "jsonrpc" : "2.0", "method" : "VehicleInfo.ReadDID", "params" : { "appID" : 1329276570, "didLocation" : [ 56832 ], "ecuName" : 2000 } } ]

From HMI to SDL
=====================
reportMessage::TO_SDL [{ "error" : { "code" : 2, "data" : { "method" : "VehicleInfo.ReadDID" }, "message" : "The Requested ECU is not Available" }, "id" : 54, "jsonrpc" : "2.0" } ]

SDL to App:
=====================
[application_manager_impl.cc:2124] :ConvertSOtoMessage::Convertion result: { "info" : "The Requested ECU is not Available", "resultCode" : "UNSUPPORTED_RESOURCE", "success" : true }

Thanks,
Nilesh

JackLivio, 2019-04-11:
@Nilesh-harman I believe this behavior is expected. In this case success: true can be explained by the fact that the message was successfully received and processed by the HMI. 

Because the interface was available and the HMI responded with UNSUPPORTED_RESOURCE, the app should read this as the HMI does not support this information. 

If the HMI interface was unavailable, success: false would be the response in which tells the mobile app it should maybe try the request again.

JackLivio, 2019-04-11:
We can make a note to update the integration guidelines for this case.

Nilesh-harman, 2019-04-17:
@JackLivio , Thank you very much for suggestion and clarifying my doubt.

JackLivio, 2019-04-25:
Add description to integration docs in [#147](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines/pull/147)

closed 2019-04-25 by JackLivio